keirf/flashfloppy #612: LED stays on and Amiga isn't properly reading floppies

therealjbenam, 2022-02-28:
Hi,

Thanks for the great work on FlashFloppy. I have bought a bunch of Goteks which I am now flashing to use with my various hardware.

Two Goteks went fine without any issue, but the third one is showing some weird issues - the read LED stays always on (even without a floppy cable!) and while it's correctly detected by the Amiga, it doesn't read stuff properly (I get a "Please insert X in DF0:" and I think I can hear noises from the speaker that resemble when real floppies have trouble reading.

Soldering is fine and top-notch - I have also removed every hardware mod and the same issue still happens. I have reprogrammed it 3-4 times with the DFU, even went back to 3.29, but nothing changes.

What could be causing this issue?

Thanks in advance for your time.
Best Regards

keirf, 2022-02-28:
Which LED is staying on unexpectedly: Green or Red?

Jumpers set correctly?

therealjbenam, 2022-02-28:
> Which LED is staying on unexpectedly: Green or Red?
> 
> Jumpers set correctly?

Hi, green LED, which I suppose it's the read one.

Just S0, I don't usually set other jumpers with Amiga.

keirf, 2022-02-28:
Then there's something electrically wrong on this Gotek board. The green LED can only light when the drive is selected. Which means the drive select line is tied to ground on this drive, even when no floppy cable is connected. Could be a blob of solder or something?

closed 2022-02-28 by keirf

therealjbenam, 2022-02-28:
> Then there's something electrically wrong on this Gotek board. The green LED can only light when the drive is selected. Which means the drive select line is tied to ground on this drive, even when no floppy cable is connected. Could be a blob of solder or something?

Sadly no shorts on any of the pins neither any solder blobs laying around. I have found that one of the 1K resistors (RN5) is out of spec at around 620 Ohm. Not sure if that could be the origin of the issue though.

What else should I check? Is there a diagnostic mode or something for the STM32 itself?

Thanks.

keirf, 2022-02-28:
Not really. Just bin it and use another one.

therealjbenam, 2022-02-28:
> Not really. Just bin it and use another one.

Welp. Last question: would it be worth it to purchase a replacement STM32? Still cheaper than an entire Gotek...

keirf, 2022-03-01:
If you can find the chip :)

therealjbenam, 2022-03-01:
True. Thanks again for the wonderful FlashFloppy.

therealjbenam, 2024-01-13:
Hello there @keirf - I was wondering if I could replace the STM32F105RBT6 with an AT32F435. From what I have read around, they should be pin-compatible with the Gotek PCBs used for the STM32. My PCB in question is a SFRC922D.

I am just a bit undecided on which sub-model I should be getting: AT32F435RCT7 with 256KB of flash, or AT32F435RGT7, with 1024KB of flash? (I don’t think FlashFloppy will make use of the RMT7 with 4032KB of flash?)

They are going for a few euros on AliExpress - worth a try, imho.

Thank you for help!

therealjbenam, 2024-01-15:
Looking at a few high res pics I have determined that the AT32F435RGT7 is the one used in the new Goteks. Bought 3, just in case. Will report back in a few weeks.