keirf/flashfloppy #613: Gotek on BBC Master

EuanWebster, 2022-02-28:
v3.30 on Artery Gotek.
BBC Master with 1770 disk interface.

When issuing *CAT, drive light comes on, but not showing contents of disk. The BBC just hangs there waiting for something to come back from the Gotek. 

FF.CFG is the same as other BBC users. Posted at end of this post.

The Log file shows this:

```
** FlashFloppy v3.30 for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Build: Jan 26 2022 08:21:20
Board: Standard
Config: Flash Slot 1 (ver 2, size 70)
Interface: Shugart (pin2=chg, pin34=rdy)
I2C: Invalid bus SCL=1 SDA=0
Display: 3-Digit LED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0123
 PID : 4567
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : GENERAL
 Product : USB 2.0 PenDrive
 Serial Number : 0000000000000311
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 42 items
Mode: Native
Cache 42 items
Current slot: 0/2
Name: 'GAMES' Type: ssd
Attr: 20 Clus: 00000008 Size: 204800

** FlashFloppy v3.30 for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Build: Jan 26 2022 08:21:20
Board: Standard
Config: Flash Slot 1 (ver 2, size 70)
Interface: Shugart (pin2=chg, pin34=rdy)
I2C: Invalid bus SCL=1 SDA=0
Display: 3-Digit LED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0123
 PID : 4567
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : GENERAL
 Product : USB 2.0 PenDrive
[... 304 lines cut ...]

# LCD/OLED long filename scroll rate during navigation (ms per update)
# Values: 0 <= N <= 65535
nav-scroll-rate = 80

# LCD/OLED long filename pause before scroll, during navigation (milliseconds)
# Values: 0 <= N <= 65535
nav-scroll-pause = 300

##
## MISCELLANEOUS

# Speaker volume for head STEP
# Values: 0 <= N <= 20
step-volume = 20

# Report the specified version number to host software
# Values: <quoted-string> ("" means report real version)
# eg. da-report-version = "v3.0.0.0"
da-report-version = ""

# Automatically extend certain types of truncated image file (SSD,DSD,TRD)?
# Values: yes | no
extend-image = yes
```

RetroClinic, 2022-02-28:
After selecting the image, are you pressing the encoder to insert it?

EuanWebster, 2022-02-28:
Hi Mark. Don't have a rotary on it yet. Just the two buttons to go up and down. I assumed that auto-inserted the disk?

EuanWebster, 2022-02-28:
Ah, thanks Mark. That's great. I'll give it a go.
I took a copy of the file for the BBC Micro off the internet, which obviously turns out to be your version.

EuanWebster, 2022-02-28:
Needed a couple more changes and now works. Many thanks.

New file below:

```
## FF.CFG: Example FlashFloppy Configuration File

# Place in the root folder or FF/ subfolder of your USB drive.
# NOTE: If FF/ exists, IMG.CFG must reside there, not the root folder.

# Uncommented lines below are the default settings.
# Uncommented options cannot be overridden by settings in other config files.

##
## DRIVE EMULATION

# Floppy-drive interface mode (interface pins 2 and 34)
# jc: Specified by jumper JC (open: shugart, closed: ibmpc)
# shugart: P2=DSKCHG, P34=RDY (Amiga, Atari ST, many others)
# ibmpc: P2=unused, P34=DSKCHG (IBM PC interface)
# ibmpc-hdout: P2=HD_OUT, P34=DSKCHG (not generally needed: prefer 'ibmpc')
# jppc: P2=unused, P34=RDY (Japanese PC standard)
# jppc-hdout: P2=HD_OUT, P34=RDY (Japanese PC alternate: prefer 'jppc')
# akai-s950: Legacy alias of 'jppc-hdout', previously used for Akai S950
# amiga: P2=DSKCHG, P34=DRIVE_ID (not generally needed: prefer 'shugart')
interface = shugart

# Host platform: Improves image-format detection for generic types such as IMG
# acorn: Acorn ADFS
# akai: Akai (S01, S20, S950), Korg, SC Prophet 3000
# casio: Casio (FZ-1)
# dec: DEC (RX33, RX50)
# ensoniq: Ensoniq (ASR, TS, etc)
# fluke: Fluke 9100
# gem: General Music (S2, S3)
# ibm-3174: IBM 3174 Establishment Controller
# memotech: Memotech
# msx: MSX
# nascom: Nascom
# pc98: NEC PC-98
# pc-dos: PC DOS Format (geometry determined from Bios Parameter Block)
# tandy-coco: Tandy Color Computer (CoCo)
# ti99: TI-99/4A
# uknc: UKNC / DVK Soviet PDP-11
# unspecified: Common default geometries (including IBM PC)
host = acorn

# Pins 2 & 34 output (drive->host) manual configuration
# auto: Auto-configure from interface= setting
# nc: Unused/No Connection [eg. if pin is an input (host->drive) on your host]
# low: Always 0 volts (0v)
# high: Always 5 volts (5v)
# rdy: Drive ready (Ready = 0v)
# nrdy: Logical complement of above
# dens: Density mode (High Density = 0v)
# ndens: Logical complement of above
# chg: Disk changed (Changed = 0v)
# nchg: Logical complement of above
# Values: auto, nc, low, high, rdy, nrdy, dens, ndens, chg, nchg
pin02 = auto
pin34 = auto

# Forcibly write-protect images, or respect the FAT read-only attribute?
# Values: yes | no
write-protect = no

# Maximum cylinder that can be stepped to (255 is required for access to
# Direct Access mode as used by imagee-selector utilities and Autoswap games).
# Values: 0 <= N <= 255
max-cyl = 255
[... 187 lines cut ...]

# LCD/OLED long filename scroll rate during navigation (ms per update)
# Values: 0 <= N <= 65535
nav-scroll-rate = 80

# LCD/OLED long filename pause before scroll, during navigation (milliseconds)
# Values: 0 <= N <= 65535
nav-scroll-pause = 300

##
## MISCELLANEOUS

# Speaker volume for head STEP
# Values: 0 <= N <= 20
step-volume = 10

# Report the specified version number to host software
# Values: <quoted-string> ("" means report real version)
# eg. da-report-version = "v3.0.0.0"
da-report-version = ""

# Automatically extend certain types of truncated image file (SSD,DSD,TRD)?
# Values: yes | no
extend-image = yes
```

closed 2022-02-28 by EuanWebster