keirf/flashfloppy #614: Updates to the "supported devices" Wiki pages

chriswareham, 2022-03-06:
I can confirm FlashFloppy works with the following devices:

- Yamaha QX1 MIDI sequencer
- Yamaha QX3 MIDI sequencer
- Yamaha SY77 synthesiser

The SY77 requires an adapter for the oddball connector, same as the SY99.

keirf, 2022-03-10:
Can these devices fit under existing Yamaha sub-headings in the wiki? eg QX5FD for some of them?

chris-wareham, 2022-03-10:
The SY99 could go under the same sub-heading as the SY77 - they are very similar. The QX3 could also go under the same sub-heading as the QX5FD, as they are also similar. The QX1 differs to the other models in the QX series though, as it had a 5.25" floppy drive rather than a 3.5" one. The QX1 also seems to use its own proprietary disk format, whereas the later models uses MSX-DOS (FAT12).

chris-wareham, 2022-03-10:
I'll pop open my QX machines this evening and check what jumper settings I'm using.

chriswareham, 2022-05-18:
Hi Keir, I completely forgot about this. Just to confirm, all of my QX machines have the jumper at S0 only.

closed 2022-12-19 by keirf