keirf/flashfloppy #619: Amiga HD ADFs not working with 3.30 firmware.

AXm77, 2022-03-11:
I tested A3000 (OS 3.2.1) with Gotek (older one) as internal FDD, flashed to 3.30 firmware and HD ADFs throwing a error ( it isn't recognized as HD floppy?). But same config working without problem (didn't test writing just reading files) with 3.13a firmware.

keirf, 2022-03-11:
Can you try intermediate versions (eg 3.22 and so on) to work out which introduces the regression?

Tallguy297, 2022-03-14:
I am using firmware 3.25 with no issues.
However I have had to put the FF.CFG & IMG.CFG files in the FF directory on the USB thumb drive.
Change line 21 from `interface = jc` to `interface = amiga`
Since then I have had no issues with reading DD & HD ADFs.

The latest firmware doesn't seem to work well with my Amiga 600 Vampire.

keirf, 2022-03-14:
So to be clear: v3.29 introduces problems on 600 vampire, and v3.28 seems okay?

Can you open a new ticket and describe the symptoms, assuming they are broader than the scope of this ticket?

Tallguy297, 2022-03-14:
Yes, I opened up a ticket.
`https://github.com/keirf/flashfloppy/issues/622`

keirf, 2022-03-15:
@AXm77 I have test v3.30 with my STM32 Gotek and 1760k ADFs are recognised fine on my fairly vanilla A500 running KS3.1/WB3.1.

Perhaps it's a speed issue. What CPU does your A3000 have?

closed 2022-03-17 by keirf