keirf/flashfloppy #622: Unusual problem with ADF files.

Tallguy297, 2022-03-14:
I recently installed a Vampire 2 card in my Amiga 600.
Since then, any firmware after 3.29 does not work as intended.
With firmware 3.29, if I insert the USB thumb drive, turn on the Amiga and select an image it will not boot.
It boots straight to the hard-drive and drive DF0: does not appear and is not mounted.

The Amiga will only recognise the ADF when I turn on the Amiga and then insert the thumb drive.
If I do a soft reset using the CTRL+L-Amiga + R-Amiga with the thumb drive inserted, nothing happens.
It just boots straight from the hard-drive.

With firmware 3.28 and earlier, this problem does not exist.
I can insert the thumb drive, turn the Amiga on and then boot from any ADF image.
If I boot from the hard-drive, as per normal behaviour, the ADF image will properly mount in drive DF0:

It's not a huge problem, but something I thought you might like to be aware of.

keirf, 2022-03-14:
Do you know the Gotek model (eg as written on the Gotek PCB)? If not then approximate age of the Gotek may be helpful.

Is the issue definitely vampire related? Do you know for sure whether v3.29 works ok with vanilla 600?

Is this tested only with `interface=amiga` in FF.CFG or has `interface=shugart` been tried?

Tallguy297, 2022-03-14:
My gotek is the older model, at least 1&1/2 years old.
I can't provide a model number from the PCB.
I have had my Amiga apart so many times now that the clips are breaking.
I am unwilling to do more damage.

Yes, I believe the issue is Vampire related. Prior to this I had a Furia accelerator and Kickstart 3.2.
The only problem I had was using a compatible CF Card for the OS. The gotek floppy drive had no issues up until firmware 3.30.

I have only had the Vampire 2.2 board for about 1 month. As soon as it was inserted I had the problems.
I have flashed the Vampire Rom with kickstart 3.2.1 and the gotek drive still plays up with any firmware later than 3.28.

I have tried the gotek with the `interface=shugart` option, but then I cannot read my HD ADF images.
Using `interface=amiga` gives me the best results.

Thank you.

keirf, 2022-03-14:
Ok thanks it is definitely an STM32 model if it's that old.

Were you trying to boot HD images and failing, or were you trying to boot DD images?

keirf, 2022-03-14:
So the suspicion here is that in supporting the latest Artery-based Goteks with QFN32 chip, the drive-select IRQ handler got more complicated and a bit slower. Perhaps it is now too slow for Vampire-based Amigas. I remember them not working properly many years ago and I had to optimise the IRQ handler carefully.

Unfortunately I no longer have a Vampire, and so fixing this will be reliant on a bit of legwork on your part: Basically trying a few things that I suggest, and also testing some trial fixes. If you're okay with this then we can keep this ticket open to track progress, and get to work. There's no deadline on this of course, just as and when you have time is fine, you just need to be willing.

One thing that would be immediately interesting is:
- Update to FlashFloppy v3.30
- Set `interface=shugart`
- Try booting DD images
 
DD images have a shorter simpler IRQ handler path, and also they don't need to "waggle" the READY line to identify an HD disk. So this configuration *might* work. If so, it shows the IRQ handler is marginal. If not, I might need to come up with a bigger idea. :)

Tallguy297, 2022-03-15:
I tried, as per your suggestion to change the configuration to `interface=shugart` and boot from a DD ADF image using firmware 3.30.
Unfortunately, no go.
It only seemed to recognise the drive when I inserted the USB after power on.
If I was quick enough to insert the USB thumb drive before the hard-drive could boot then it would work but most of the time is would boot from the hard-drive into the OS.

keirf, 2022-03-15:
Thanks for the test. That's the same as you observed using `interface=amiga` isn't it? I'll have a think...

keirf, 2022-03-15:
And it is DF0 right? Extra weird if so as the Kickstart assumes DF0 always exists and shouldn't need to positively identify it. Argh.

Tallguy297, 2022-03-15:
If I use `interface=amiga` then I can use HD ADF images. If I use `interface=shugart` then that is not the case.
My gotek drive is inside the amiga. I do not have any external floppy drives or additional gotek interfaces.

keirf, 2022-03-15:
Yes, all I mean is: `interface=shugart` didn't help in any way with DD images. They are broken in the same way regardless of your `interface` setting?

Tallguy297, 2022-03-15:
Yes, that is correct.

keirf, 2022-03-16:
Okay so sticking with `interface=amiga`. It doesn't boot when the USB is inserted at cold power on or even after warm reset (when a disk was previously mounted just fine in Workbench), right?

What if you leave the USB inserted, boot from HD to Workbench: DF0 does not appear. But what if you now eject the image (or, equivalently, navigate to a different image on the Gotek) and mount/insert/re-insert the image on the Gotek. Will DF0 now appear in Workbench, or do you actually really require a full USB stick physical eject/insert cycle?

Tallguy297, 2022-03-16:
In answer to your question `What if you leave the USB inserted, boot from HD to Workbench: DF0 does not appear. But what if you now eject the image (or, equivalently, navigate to a different image on the Gotek) and mount/insert/re-insert the image on the Gotek. Will DF0 now appear in Workbench, or do you actually really require a full USB stick physical eject/insert cycle?`
The drive disappears completely. It's just a strange behaviour.

However, please do not put too much effort into solving this problem. I am using firmware 3.28. I am mounting images, booting etc. and having no problems.

Computers like the Amiga have old chipsets and memory.
There is bound to be problems dealing with new hardware and configurations. I am extremely happy with the job you have spent a lot of time on help with the Gotek floppy drive emulator.
It is not an easy task and I wish you all the best.

However, my system works and that's that. Thankyou for all your continued support.

closed 2022-03-16 by Tallguy297