keirf/flashfloppy #623: *FAT*02* in OLED display and ADF not loaded

Papalapaa, 2022-03-14:
I was succesfully using my Gotek Flashfloppy 3,23 together with my Amiga 500 time ago and now I started to use them again, but when I try to load most of the games into the USB, the OLED display changes from ADF to *FAT*02* and nothing happens. Some files work but another ones not.

I tried to check/edit the HXCSDFE.CFG with the FF-HxC software Editor but the slot list isn't loaded. If I open the HXCSDFE.CFG in the Windows notepad I can see the slots list and some other characters. Is there a way to repair the HXCSDFE.CFG file, even manually?

I have more than 800 slots asigned so, to reasign again the slots one by one is the worst option...

Thank you!

keirf, 2022-03-14:
Why do you want to use slot assignments rather than direct navigation of image files via the oled display? If you remove HXCSDFE.CFG then FlashFloppy will automatically list all available image files for you to navigate.

Papalapaa, 2022-03-14:
Oh! I was not aware that to do this was possible. I was thinking that HXCSDFE.CFG was mandatory. I'll try...
Many thanks!