keirf/flashfloppy #626: Can Flash floppy be used without image software

stevehdj, 2022-03-22:
I have flash floppy usb drive, firmware ver. 3.28, installed in a Roland AT80SL and wondering if I can partition a USB stick to load and read native files without having to use an emulator software to create .hfe or other type of image.  I have read that this can be done with HcX but wish not to try and change to the HcX firmware in existing drive or purchase a new drive with the Hcx.  Thanks

Occupied, 2022-05-08:
That is pretty cool that HxC finally supports Image-less mode like the original Gotek firmware does!
I agree it would be a very nice addition to have this in FlashFloppy too!
https://hxc2001.com/docs/gotek-floppy-emulator-hxc-firmware/pages/emulation-from-partition.html

keirf, 2022-06-22:
Yes emulation from FAT12 partitions seems quite doable. I wonder if anyone cares much about emulation from folders? Doable but a bit tricksier to sync between folder and on-the-fly disk image.

Occupied, 2022-06-24:
I think emulation from folders was just a stopgap before HxC implemented the partition version, I think partitions makes more sense as then you don't risk allot of issues, and the user does not need to keep track of not exceeding the partition size manually.

One issue to keep note of is that Windows versions before Windows 10 1703 (Redstone 2) do not support multiple partitions on removable media. This includes Windows 10 LTSB versions and of course Windows 8.1 Windows 7 and older...
This can be gotten around by using USB sticks that show up as "fixed disk" in Windows but they are kind of rare.
I found one but it is kind of expensive, but it could be helpful if users make a list of USB devices in this case: Corsair Flash Voyager GTX 128GB USB 3.1

Gkiokan, 2023-04-23:
> Yes emulation from FAT12 partitions seems quite doable. I wonder if anyone cares much about emulation from folders? Doable but a bit tricksier to sync between folder and on-the-fly disk image.

Is this actually into the current FW implemented? I couldn't find any notes about it. Just the Image reference sofar which is still a extra step for creating the filled images with the files and then copy to the usb.  

It would be more convinient if we could use a USB drive directly to fill in the files per folder / disk and it reflects them on the specific drive.

Gkiokan, 2023-04-24:
I just realize this is completely OT what I want. The later one you said. 

I mean like instead of having IMG files it would be great to have folders in one partition with the files in it that would be emulated.  
This would kill the need to create each disk as IMG file and would be more convinient.  

The Problem with the partitions I see is that there is a need for third party dependency like Software like Batch Manager Tool to manage the contents of each disk/partition (which currently works only on Windows and mac users fall down).

keirf, 2023-04-24:
There are pre-made images available from the HxC project: https://hxc2001.com/docs/gotek-floppy-emulator-hxc-firmware/pages/emulation-from-partition.html

On mac these can be written using balena (or dd from a Terminal, as you would on Linux).

CFD2, 2024-01-28:
This feature would be fantastic to have for any old sampler/workstation. Typically I avoid using a Gotek whenever I can just because I have to go through the inconvenience of finding images online which is not always possible.