keirf/flashfloppy #631: AKAI S3000XL Board SFRKC30.AT2

brianrudie2, 2022-03-25:
This board seems temperamental 
 it reads OK but writing to floppy is temperamental ( 3.30 )
Jumper on S0 
Rotary, OLED

keirf, 2022-03-25:
Okay as on prev ticket, try v4.4a for starters.

brianrudie2, 2022-03-26:
Got round to upgrading it but it did not fix the issue, any ideas?

keirf, 2022-03-27:
Try another usb drive.

chrispoacher, 2022-03-27:
Also make sure the ribbon cable is fully pressed home, they are notorious on Akai samplers.

brianrudie2, 2022-03-27:
I have tried all of the below and they have not fixed the issue:

Multiple brands and sizes of USB sticks, I have  even tried ones that work
perfectly on my other samplers without issues
Inserted cable firmly, and even tried it the opposite way round ( and it
reports it the wrong way around on the OLED to confirm )
Updated the firmware to 4.3a
I have tried the jumper On S1,S0  ( In the past JC and S0 together have
always worked perfectly, but this particular model does not have the JC
jumper connector )  so if there is a fix for this it may help ?
I have also tried 3 other gotecks ( in this same model SFRKC30.AT2  ) that
I have here and they are all giving exactly the same issue.

To date, I have configured 20 +  gotecks with flash floppy
(non-artery versions ) and not had an issue until now

Any ideas?

brianrudie2, 2022-03-28:
Update: If I format the disk image on another sampler with another working gotek ( Non artery chip version ) 
It saves perfectly until  I try to do a wipe save over the disk on the sampler which results in a  write error and the save fails

keirf, 2022-03-28:
Format is harder than normal saves as the latter usually happen sector by sector so less buffering required.

brianrudie2, 2022-03-28:
Any ideas for a solution? Im struggling

keirf, 2022-03-29:
If you want an alternative to jumper JC you need the following line in FF.CFG:
```
interface = ibmpc
```

brianrudie2, 2022-03-29:
> If you want an alternative to jumper JC you need the following line in FF.CFG:
> 
> ```
> interface = ibmpc
> ```

No luck , I already tried this and got the same issue

keirf, 2022-03-29:
Then you're SOL. You should try an STM32 Gotek in this very machine.

keirf, 2022-03-29:
If you buy now you'll probably get SFRKC30.AT4 which uses the 64 pin artery chip. Still the same chip inside though unfortunately. The STM32 models aren't being made. Boards with a new artery chip with more ram are probably a month or so away.

One advantage of the SFRKC30.AT4 is that I had to make significant changes in v3.29 to support SFRKC30.AT2. With the AT4 board you can try (slightly) older firmwares. It's not much to go on though.

It might also be interesting to know if hxc firmware works better (but you have to pay for that).

brianrudie2, 2022-04-01:
Does anyone have a good place to buy these wholesale, where I know exactly which model I am receiving?  ( apart from Ali express ? )

keirf, 2022-04-01:
Email Yan at guiie@163.com, he sells wholesale direct from factory and can tell you what board you're getting. They will all be AT32F415xxx though just differing in package pin count.

chrismkz, 2022-04-07:
hi 

For the s3000 the cd 3000  I just use the hcx file for the 3200 I found on internet , surely on torlus , got the file on my computer if needed . jumper S0 and bam it works.

keirf, 2022-06-22:
New 435 based Goteks available from factory within next week. I recommend to try one for this temperamental write issue.

closed 2022-06-22 by keirf