keirf/flashfloppy #635: Roland S-760 Keeps rebooting at OS load initiating

andr229k, 2022-04-09:
**Environment:**
Roland S-760 with Gotek AT model and firmware version 3.30.
Jumper = S0
Interface Mode = ibmpc-hdout
Host = unspecified

**Diagnose:**
This is how a boot is supposed to look like -> [https://www.youtube.com/watch?v=3ftJ4UKiBjc](https://reverb.com/onward?author_id=1429259&to=https%253A%252F%252Fwww.youtube.com%252Fwatch%253Fv%253D3ftJ4UKiBjc)

In reference to following article was 3.9a [unstable version at that time] working fine with Roland S-760 with IBMPC-HDOUT as Interface mode -> http://llamamusic.com/s50s550/FlashFloppy_GOTEK.html?fbclid=IwAR1X1_X5IB1I0iuMj0_49dx_93CT2zy2NyD077rNlfJH-nrAic9EWDgcNfU

**Symptoms:**
Keeps rebooting at OS load initiating

See Video:

https://user-images.githubusercontent.com/52201354/162589808-bf1fc7dc-46d2-42ff-9830-a1a2abe409bc.mp4

Any idea/suggestion, what might keep looping?

keirf, 2022-06-22:
Did you ever resolve this? New 435-based Goteks with loads more SRAM are available in the next week or two. You should get one and try this again.

keirf, 2022-06-22:
They are not default until 435 drops in cost (currently $2 more approx) by end of year, then yes all Goteks are likely to be using 435.

closed 2022-06-22 by keirf