keirf/flashfloppy #637: Some games of sharp X1 have image errors.

SoukinOKATU, 2022-04-13:
I now use the firmware of 4.4a and use it to sharp X1 turbo.The interface has used Shugart and jppc respectively
But there are two strange questions:
1.ys1, occasionally after the title, the game freezes at the beginning (the sound will get stuck)
2. Space Harrier, I have two kinds of images, 2D and D88, but the two images are normal in the emulator. However, on x1turbo（Convert to HDF format）, the title screen lacks the option menu, and after entering the game, the whole picture is wrong. There is no music but sound effect
Images of both games can be found in "Neo Kobe sharp X1" on archive.org,And they can all run on the emulator
Here are the images of the two games
[X1.zip](https://github.com/keirf/flashfloppy/files/8479751/X1.zip)
.

keirf, 2022-04-13:
Is there any difference in behaviour using v3.30 firmware?

keirf, 2022-04-13:
And indeed are you sure these titles should run 100% on your hardware eg with real floppy drive.

SoukinOKATU, 2022-04-13:
> 

It can be confirmed that ys1 using a real floppy drive is normal (yes, I have purchased a floppy disk)

But I only have a virtual image, and this image seems to be encrypted and cannot be copied into an empty floppy disk

SoukinOKATU, 2022-04-14:
Previously, a Japanese researcher saw that using the hxc floppy disk emulator, Space Harrier title screen  is indeed normal (with menu options)
[image]
The screen running with the emulator is also normal.But if it run with gotek, the option text in the middle will disappear.
[image]
In the emulator, the normal game is like this,But if it run with gotek, The picture will turn white except for some enemies and main characters
[image]

keirf, 2022-06-22:
May I ask what Gotek type? Artery or STM chip?

keirf, 2022-12-19:
Do you still wish to pursue this issue?

closed 2023-01-17 by keirf