keirf/flashfloppy #640: Amiga 1200 Gotek as DF0 not booting

StevenHachel, 2022-04-14:
hey,

When inserting the USB stick after loading, the Workbench 3.2.1 can see the selected ADF and change the folder on mine
USB stick via Gotek step keys.

When I reset my Amiga, the Gotek does not boot. but if I remove the USB stick from the Gotek and then reset the Amiga, the Gotek (speaker) clicks. After resetting my Amiga (can start Workbench manually via the startup menu), I plug in the USB stick,
the last selected ADF is shown in the display.

When I boot up the workbench, the ADF can also be seen under the workbench.

After the reset (with USB plugged in) everything as usual. Workbench starts and ADF is no longer visible.

Can I even boot from the Gotek?

Firmware used: V3.30 and 4.3a.

keirf, 2022-04-14:
So, in Workbench, can you change disks on Gotek and that is reflected in the DF0 disk view in WB?

This could be a Shugart vs PC interface confusion. ESCOM 1200 boards are wired to expect PC drive interface.

keirf, 2022-04-14:
So what happens instead of floppy boot? HDD boot?

keirf, 2022-04-14:
And what model is your Gotek (written at rear of PCB, eg SFR....)

StevenHachel, 2022-04-14:
> And what model is your Gotek (written at rear of PCB, eg SFR....)

GOTEKsystem
SFRKC30.AT2

StevenHachel, 2022-04-14:
i have installed Eject Button (JA). When floppy ejected, klicking the gotek and can loading ADF file over the button and shows on WOrkbench. Then Reset and not booting. Eject the ADF (over button) is the Gotek not clicking. 

Only after a reset, but no ADF (simulated) must be inserted, does the Gotek drive click again and inserting and ejecting behaves like a real drive.

keirf, 2022-04-14:
Two things to try in FF.CFG: `interface=amiga` and alternatively `interface=ibmpc`. If neither help then revert to `interface=shugart`

StevenHachel, 2022-04-14:
>

StevenHachel, 2022-04-14:
> Two things to try in FF.CFG: `interface=amiga` and alternatively `interface=ibmpc`. If neither help then revert to `interface=shugart`

`interface=amiga` and `interface=shugart` | have the same behavior
`interface=ibmpc `| The drive is behaving strangely and the system says it cannot be recognized

not that my motherboard is damaged. :(

StevenHachel, 2022-04-14:
I will use the Gotek as the DF1 and put my old floppy drive back in.

StevenHachel, 2022-04-14:
> > Two things to try in FF.CFG: `interface=amiga` and alternatively `interface=ibmpc`. If neither help then revert to `interface=shugart`
> 
> `interface=amiga` and `interface=shugart` | have the same behavior (e.g. With an ejected ADF and via the early-boot menu, an ADF can be loaded and started from DH0. very confused `interface=ibmpc `| The drive is behaving strangely and the system says it cannot be recognized
> 
> not that my motherboard is damaged. :(

AmiUpi, 2022-05-04:
Hi.  Has the problem already been solved?  My gotek shows the same errors.

keirf, 2022-05-04:
Exactly same behaviour? Very odd. No it has not been worked out as yet.

keirf, 2022-05-04:
I advise to run Amiga Test Kit and run the floppy signal test on the Gotek. I can give help on expected signals when eject/insert/step if you get to that point.

StevenHachel, 2022-05-04:
> Hi. Has the problem already been solved? My gotek shows the same errors.

No, I can't get it to work.  I'm still testing and I'll get back to you in a few days.

keirf, 2022-05-04:
If your chip is AT32F415RBT7 (LQFP64 chip) then we can check if this regression was caused by adding support for the new SFRKC30.AT2 board (with tiny QFN32 chip). Here is a build, which is basically v3.28 with added support for Artery chips on Goex boards:
https://github.com/keirf/flashfloppy/suites/6369573220/artifacts/230333808

You will need to be logged in to Github to use the above download link. This firmware will not work on SFRKC30.AT2 boards.

keirf, 2022-05-04:
The blanking is strange. Is it isolated perhaps to the newer OLEDs? Can you try an old OLED in new GOEX or vice versa?

keirf, 2022-05-04:
Regarding your A1200 setup, can you confirm that apart from new ROM, it is a vanilla standard unexpanded A1200? I also wonder if this is limited to the latest Kickstart... since original report is 3.2.1 as well. I may need to buy a ROM image for testing purposes.

keirf, 2022-05-04:
I will have to dig out my 1200 and repro the boot issue. I've ordered OS 3.2 also; it's handy to have.

No report before on the OLED issue. It may be OLED unit specific. It may be SD specific. Perhaps relating to wire noise. If need be I can do you a patch to try which allows slowing the I2C to standard rate. Possibly fiddling with pullup strength on the I2C lines might help too.

keirf, 2022-05-04:
The i2c is constantly streaming. It recognises when things go wrong and will automatically reset the display. That's the blanking you see. You may also get error output logged. In which case it is worth trying the logfile firmware and collecting a FFLOG.TXT file (https://github.com/keirf/flashfloppy/wiki/Logging-To-USB-Drive). You would exhibit the OLED problem, and then mount an image and eject it, and then that should cause logging to get synced onto the SD card.

keirf, 2022-05-04:
Definitely I2C badness (timeouts, stuck comms) for some reason. Here's a new version of your special firmware which runs the I2C bus at a slower rate (standard rate, which is 4x slower than fast rate, to be precise):
https://github.com/keirf/flashfloppy/suites/6374914142/artifacts/231021144

keirf, 2022-05-04:
I'll have a think whether to make oled always slower, or add a config option.

Regarding the boot issue, I wonder if it's related to drive detection on pin 34. Could you update back to v3.30 and try the following lines in FF.CFG 
```
interface=shugart
pin34=nc
```

AmiUpi, 2022-05-04:
So just did some tests.  Namely an Amiga 2000 with an older Gotek and an older FlashFloppy firmware.  Runs naturally.
 Then an Amiga 500. Freshly restored, with a new Gotek.  Unfortunately, I accidentally bought the Gotek with the HXC firmware.  It works too.
 Now a Commodore Amiga 1200. Freshly recapped with Apollo Icedrake and a new Gotek.  The Gotek which causes problems.
 Have now built the Gotek from the A500 in the A1200.  And lo and behold?  The drive that works in the 500 also causes problems here.
 Then installed the old Gotek from the A2000 in the A1200 and it works perfectly as it should.
 Last but not least, the new Gotek built with FlashFloppy in the Amiga 2000.  Oddly enough, the drive also works perfectly here.
 Everything tested with a USB stick.

 So the preliminary conclusion.  The new Goteks are causing problems in my A1200, regardless of whether they are HXC or FlashFloppy.  Both drives run smoothly in the A2000 and A500.

 The old Gotek runs perfectly in my A1200.

 Now I'm curious who solves the riddle.  I think there are some chip components that don't get along.
 I ran this through google translator.  I hope it is understandable.
 Best regards

keirf, 2022-05-04:
@AmiUpi That's interesting, potentially your A1200 is tougher as it has a Vampire accelerator. They were tough before, to get the pin-34 signalling (used for drive detection) fast enough. What Kickstart version is your 1200 running? I'm going to guess 3.2.1 again. I feel that ROM 3.2 has maybe changed drive detection for DF0 and it's less forgiving. I can't be sure though.

@MC6809E089 This is useful information. It tells me this must be to do with the pin-34 signalling. I guess I made it too slow, the Amiga reads a random Drive-ID bitstring, and Kickstart 3.2.1 rejects it and decides there's no DF0. So I'm going to have to reproduce this and work out how on earth I can fix it.

keirf, 2022-05-04:
@MC6809E089 Ah, but it must be more complex than that, because the STM32F105 based Gotek running v3.30 and without the FF.CFG workaround still works fine on your A1200, right? So that's a bit weird...

keirf, 2022-05-04:
The most significant test so far is confirming that v3.30 with `pin34=nc` works on your A1200. And, conversely, if you change that line to `pin34=auto`, it should fail again just as it did originally. And if you set it to `pin34=low` you should find that your Amiga never recognises the drive, regardless of whether or not an image is mounted at boot time.

If all the above is true, it really has to be related to the drive-id on pin 34. Amiga OS doesn't use the pin 34 for its regular READY signal purpose at all. The behaviour is what I'd expect if the Amiga tries to detect DF0 during boot, fails, and decides there is no DF0. Thereafter it ignores DF0: No drive click, and no disk insertions will be detected.

Why could it differ on A1200? Timing. For a start the 68EC020 is clocked twice as fast and has an instruction cache. So the drive-id loop can run faster. It will still be limited somewhat by access time to the CIA chip connected to floppy pin 34. But even that could differ a bit from A500 and perhaps be a bit faster.

Why could it be okay on old STM32 Gotek/GOEX? Timing perhaps, or electrical signalling speed. AT32F415 is a STM32F105 clone, but it won't behave absolutely 100% the same. Perhaps it's even a different component used elsewhere on the newer GOEX boards (unlikely, but can't be ruled out). Probably the drive-select ISR is just running a bit slower than on STM32F105, for some weird reason. This is critical: The Amiga drive-id loop selects and deselects the Gotek drive in a tight loop, reading the pin 34 level each time. The Gotek's drive-select ISR has to re-enable the pin 34 signal quickly enough that the Amiga reads the correct level every time.

The good news is there are a few things we could try to speed things up:
1. I can set a faster slew rate on floppy GPIO pins. This may swing them low, to 0v faster. It wouldn't help on upswings though (the signals are open collector and pulled up by resistors).
2. I could put the MCU exception table in SRAM rather than Flash. This could save a few cycles entering the drive-select interrupt handler.
3. Since this is affecting AT32F415 only, I coudl actually double the AT32F415 clock speed. It currently runs at 72MHz, which is the STM32F105 limit. But AT32F415 can run at up to 150MHz. This would obviously speed up the ISR.

keirf, 2022-05-04:
Starting with option (1), here is v3.30 firmware but with faster slew rate (actually specified as a higher source/sink strength in AT32F415 documentation). There are no other changes. It will be interesting to know if this affects boot behaviour or OLED behaviour on the problematic Gotek, with `pin34=auto`.
https://github.com/keirf/flashfloppy/suites/6379667502/artifacts/231524217

AmiUpi, 2022-05-05:
> @AmiUpiDas ist interessant, möglicherweise ist Ihr A1200 härter, da er einen Vampir-Beschleuniger hat. Sie waren zuvor schwierig, um die Pin-34-Signalisierung (die für die Laufwerkserkennung verwendet wird) schnell genug zu bekommen. Welche Kickstart-Version läuft auf deinem 1200? Ich tippe wieder auf 3.2.1. Ich habe das Gefühl, dass ROM 3.2 vielleicht die Laufwerkserkennung für DF0 geändert hat und weniger nachsichtig ist. Ich kann mir aber nicht sicher sein.
> 
> @MC6809E089Dies sind nützliche Informationen. Es sagt mir, dass dies mit der Pin-34-Signalisierung zu tun haben muss. Ich glaube, ich habe es zu langsam gemacht, der Amiga liest eine zufällige Laufwerk-ID-Bitfolge, und Kickstart 3.2.1 lehnt sie ab und entscheidet, dass es kein DF0 gibt. Also muss ich das reproduzieren und herausfinden, wie ich es beheben kann.

I actually have the roms 3.2.1.  Removing the vampires doesn't change Gotek's behavior.

keirf, 2022-05-05:
@AmiUpi which Gotek PCB versions are causing problems for you? Are they running v3.30 firmware?

AmiUpi, 2022-05-05:
Firmware is V3.30.  How can I detect PCB?  The Gotek is screwed and new from the Amigastore.  But I can open that later.  Unfortunately I have to go to work now.

keirf, 2022-05-05:
It is written at rear of the PCB. It is visible even with the case on.

AmiUpi, 2022-05-05:
[image]

AmiUpi, 2022-05-05:
[image]

AmiUpi, 2022-05-05:
SFRKC30.AT2

keirf, 2022-05-05:
Okay so the common factors again are Artery chip, A1200, Kick-start 3.2.1

TheFrenchy75, 2022-05-05:
> So the preliminary conclusion. The new Goteks are causing problems in my A1200, regardless of whether they are HXC or FlashFloppy. Both drives run smoothly in the A2000 and A500.

It appears that a new HXC firmware was released a month ago to fix a quite similar issue :

`- (7 April)  : QFN32/SFRKC30.AT2 Gotek :
Low level IO routines optimized : Faster floppy signals update.
(Fix possible issues on accelerated Amiga and fast machines)`

https://hxc2001.com/download/floppy_drive_emulator/USB_HFE_hxc_floppy_emulator_firmware_release_notes.txt
https://hxc2001.com/download/floppy_drive_emulator/HXCFEUSB_Gotek_HxC_firmware.zip

keirf, 2022-05-05:
@MC6809E089 Did you reset to `pin34=auto` in FF.CFG? I'm dubious as a I still see `pin34=high` in FFLOG.TXT which indicates that pin 34 is still "nobbled" from previous tests. If you want to do a clean test and weren't using FF.CFG for other reasons you can also always:
1. Delete FF.CFG from your Flash device
2. Hold both buttons (or rotary) with no Flash device inserted, until the display changes to `Reset Flash configuration`

keirf, 2022-05-05:
Ah, ok, yes that's fine. The pin34 handling is a bit hacky when `interface=amiga`.

Well, this slew rate hack is pretty easy if it really does generally work!

@AmiUpi and @StevenHachel can you also try this test firmware?
https://github.com/keirf/flashfloppy/suites/6379667502/artifacts/231524217

@AmiUpi also interesting if it works with and without Vampire.

keirf, 2022-05-05:
@MC6809E089 Here's another test firmware which tries dropping the slew/drive on the SD SPI pins. Maybe it will help with your OLED issue. Otherwise it is identical to the previous test firmware (ie. the boot issue should still appear fixed):
https://github.com/keirf/flashfloppy/suites/6385669293/artifacts/232016144

keirf, 2022-05-05:
Okay, so it seems now we need just the dev tip of v3, plus a fix/workaround for your OLED issue (which I will sort out asap).

@AmiUpi and @StevenHachel can you please try this (dev tip of v3) and see if it solves all your Gotek boot problems on A1200?
https://github.com/keirf/flashfloppy/suites/6369662415/artifacts/230339350

keirf, 2022-05-05:
@MC6809E089 can you try the following test firmware which is dev tip of v3 plus a fix for the OLED. To activate this fix (workaround) you add `-slow` to your `display-type` configuration. For example: `display-type = oled-128x64-slow`.

https://github.com/keirf/flashfloppy/suites/6389934250/artifacts/232305857

StevenHachel, 2022-05-05:
When I get my GOTEK back I look forward to testing the test firmware. :-)

thanks for the work!

keirf, 2022-05-09:
Any further testing from anyone lurking on this ticket? This is the build to test, on the stable v3 branch:
https://github.com/keirf/flashfloppy/suites/6405904031/artifacts/233345992

keirf, 2022-05-09:
@MC6809E089 Thanks that will be useful, though I was hoping to flush out another tester or two!

AmiUpi, 2022-05-09:
Hi.  Sorry I haven't been able to test this yet.  Didn't have time at the moment.  I'll make up for that soon.  Many thanks for your help.

bkodenkt, 2022-05-19:
I can confirm setting

pin34=nc
interface=shugart

as suggested in https://github.com/keirf/flashfloppy/issues/640#issuecomment-1117589112 makes the drive work as expected with regular FlashFloppy 3.3.

I then tried the test build as mentioned in https://github.com/keirf/flashfloppy/issues/640#issuecomment-1120851455. This works with pin34 set to auto again.

Nice!

keirf, 2022-05-19:
Thanks, that's enough to trigger a 3.31 release this weekend.

closed 2022-05-27 by keirf

StevenHachel, 2022-06-07:
@keirf many thx for this work! Gotek work with Firmwae 3.32 and Vampire accellerator and AOS 3.2.1. :)
I'm really happy that it works now!

Steven

keirf, 2022-06-07:
> @keirf many thx for this work! Gotek work with Firmware 3.32 and Vampire accellerator and AOS 3.2.1. :) I'm really happy that it works now!

Great to hear, thanks! There will be a 3.33 release in a week or so, adding support for a new Gotek CPU type. Please keep an eye out and test that one too after release, when you have time!

StevenHachel, 2022-06-07:
I will do that! :)

keirf, 2022-06-24:
Great to hear, thanks for testing!