keirf/flashfloppy #641: Issue with v3.30 and directories.

SmoothAdmin, 2022-04-17:
I tried this on an existing flash drive with Gotek adf files for my Amiga, and an a fresh new drive (I bought a 20 pack).

When I create a folder called OS, and put 3 folders in it, 3.1.4, 3.2, and 3.2.1 Update and copy the appropiate files in them when I get into the Gotek menu and access OS -> 3.1.4 I see the files. When I access OS -> 3.2.1 Update I also see the files, but when I try to access OS -> 3.2 the screeen goes blank for a sec then takes me back to the OS folder. 

I checked the flash drive on my PC and the files are there, so I copied the files from OS -> 3.2 and put them into the root of the flash drive and inserted back into the Gotek, I can see all the files on the root of the GoTek drive and they all work.

So I am not sure why the Gotek menu software reject a folder called 3.2 ?? 

Thanks

keirf, 2022-05-11:
I will try this folder setup myself and see what happens.

keirf, 2022-05-14:
Works for me. Closing. Reopen if you can provide a better repro scenario. For example, if you can image your USB stick's filesystem and zip it up and upload it somewhere so I can see exactly what the FAT filesystem looks like on your stick at the block level.

closed 2022-05-14 by keirf