keirf/flashfloppy #642: FF & OS/3 1.3

hougaard, 2022-04-17:
Sorry, this is an incomplete issue description, but I thought I would at least write down what I know. Maybe someone has a clue how to fix it :)

Machine: IBM P70 8573-121 - Running IBM OS/2 version 1.30

Issue: Disk read/write results in "Sector not found errors" - not always, but quite often when accessing the gotek with FF 3.30 from IBM OS/2 version 1.3

If I boot the machine to DOS everything works as expected, but accessing the images files that works with DOS fails under OS/2.

I can format a floppy from OS/2, but when the formatting is complete, some operation gives the sector not found error.

keirf, 2022-04-17:
Try v4.4a, try another usb stick especially if your is unbranded. Try `track-change=realtime` in FF.CFG. Not necessarily all at once.

hougaard, 2022-04-17:
Thanks, I will do that.... Menstick is a SanDisk... so pretty branded :)

keirf, 2022-04-17:
Also what format and size are your disk images and what model is your gotek (written at rear of pcb eg sfrkc30.at2)

hougaard, 2022-04-17:
It's an AT2, and all disks are 1.44mb as .IMG files.

hougaard, 2022-04-17:
- 4.4 just produced garbled output from "dir". Did see different behaviour between a formatted image, and an unformatted image, but just in error messages
- track-changes, no difference on both 3.3 and 4.4

hougaard, 2022-04-17:
Even running 4.4 when the machine is booted in DOS (where 3.3 worked perfectly) produces garbled output.

keirf, 2022-04-17:
Not a positive result for 4.4a then!

keirf, 2022-04-17:
I'll have a think about 3.30. Perhaps a gotek with more ram would help. The AT2 board chip has only 32k ram. Half what the old stm32 based goteks have.

hougaard, 2022-04-17:
Same result with 4.3 and 4.2 (just for verification)

hougaard, 2022-04-17:
> Perhaps a gotek with more ram would help. 

So, whitch one is the best to buy? It's complete minefield out there?

hougaard, 2022-04-17:
... another update:

Actually 3.30 is not perfect when booted to DOS, reading is perfect, but writing fails if I try copying anything larger than 4KB. (my initial write test happens to be a 4KB file).

I can copy multiple files to the disk, but as soon as the file is larger than 4kb it fails.

keirf, 2022-04-17:
All current goteks are the new type. Stm32 is only available if you are lucky to find NOS or a seller explicitly selling the old type. Soon I should have boards with another new chip with 384kb ram. Perhaps a month's time.

keirf, 2022-04-17:
Regarding the DOS error did you switch back to `track-change=instant` after the failed experiment? Or delete FF.CFG and reset cached config by holding both buttons for a few seconds while no usb stick is inserted.

keirf, 2022-04-17:
Another possibility could be bad hardware connection. It's pretty hard to get DOS to complain about reads and writes since writes for example are sector at a time so buffering isn't a problem, and the next sector won't be started until the sector is found via read, and timeouts are based on seeing index pulses which gotek isn't generating while the previous sector is written back to usb stick. Unfortunately its hard for you to see if this is a firmware regression since the AT2 board is supported only since 3.29!

hougaard, 2022-04-17:
> Regarding the DOS error did you switch back to `track-change=instant` after the failed experiment? Or delete FF.CFG and reset cached config by holding both buttons for a few seconds while no usb stick is inserted.

I'll verify that after lunch....

hougaard, 2022-04-18:
... lunch turned into the evening (had a video to record), no luck, track-change is back, same result. As soon as I'm writing more than 4k, it fails. 

(Tried with 3.29 also, same result).

keirf, 2022-04-18:
No further ideas regarding firmware I'm afraid.

hougaard, 2022-04-18:
OK, got my hands on a SFRC922D with the STM32F105 CPU, will try that later today...

keirf, 2022-05-11:
how did that go?

hougaard, 2022-05-11:
Sorry, meant to write a longer follow-up, life got in the way...

Short version, no-dice - something is still not allowing write operations over 4kb, also on the STM32 with 4.4 :(

I'm currently trying to locate a replacement drive (and also trying to get the current drive re-capped) to at least exclude other pieces of this old PC to be the problem...

keirf, 2022-06-22:
Try with 3.33 and the STM Gotek?

keirf, 2022-07-26:
I can also recommend trying a 435-based Gotek if you can get your hands on one.

ltning, 2022-10-03:
Just an aside - I have a 286 motherboard here which seems to eat floppies for lunch (and any other meal). Reading is *usually* OK for a little while, but it will not get far beyond booting a simple configuration. After that, your floppy or image might be in trouble. Trying to write anything at all is a guaranteed death knell for the media in use. 
This applies to FF images, real 3.5" and real 5.25" drives. I've tested with different controllers, and the most conservative BIOS settings I can. I'll be testing with another (the third!) set of RAM sticks, but at this point I suspect the motherboard is simply bad.

closed 2022-12-19 by keirf