keirf/flashfloppy #645: Cant save files from Roland AT80SL internal memory to USB floppy emulator

stevehdj, 2022-04-27:
I have a Gotek USB floppy emulator with latest version of FF firmware  installed on an Roland AT80SL organ and having problems saving from organ to USB drive?.  I can read all file types (stl,reg,midi)  into organ memory from the USB drive but cannot save any files from the organ back to USB drive.   Is this normal , I am not sure if  the AT80SL is able write to a .hfe image file?  Is there anyway to configure the drive to not have to use .hfe image files and read/write native Roland files?

keirf, 2022-04-27:
Roland usually uses PC disk format so quite likely your HFE can be converted to IMG files, either 720k or 1.44M in size.  These would be easier to write as the images are smaller. If you don't know how but have an HFE to share I can advise.

stevehdj, 2022-04-27:
Hi,, I have converted about 100 HFE files to IMG using the batch convertor,  very simple process and now am able to read and write files between organ and USB drive.  Thanks very much for your solution.

closed 2022-05-11 by keirf