keirf/flashfloppy #650: Amiga ADF HD Image problem with firmware 3.31

Tallguy297, 2022-05-25:
I cannot boot any ADF HD image since upgrading to firmware 3.31.
ADF DD images boot OK.

I have degraded the my Gotek firmware to version 3.28.
Booting from either ADF HD or DD image works fine with this version.

Regards,
Alexander.

keirf, 2022-05-25:
Do versions 3.29 or 3.30 work?

keirf, 2022-05-25:
Also what is your Amiga setup?

Tallguy297, 2022-05-25:
Versions 3.29 & 3.30 do not work. This issue was raised and closed in #619.
I have an Amiga Vampire 2.2, Kickstart 3.2.1, and the Gotek is the older chipset.

keirf, 2022-05-25:
Okay, here is a test firmware which is 3.31 but with the interrupt handler partially reverted to 3.28. This simplifies it and makes it quicker to run. If this fixes your issue, we can work out exactly how far the handler needs to be simplified, and sort a proper fix for 3.32. You need to be logged in to GitHub for the link below to work:
https://github.com/keirf/flashfloppy/suites/6649852846/artifacts/251268051

Tallguy297, 2022-05-25:
Thanks for the firmware update. It partially worked.
Rebooting into ADF DD images, again no problem.
Rebooting into ADF HD images, the floppy led light stayed on, the Gotek drive stuck on track 0, and I received a DF0: disk error message.

keirf, 2022-05-25:
I'm amazed it worked as well as it did. There were a couple of bugs. That code is really fragile. Please try this one instead:
https://github.com/keirf/flashfloppy/suites/6651124084/artifacts/251342494

Tallguy297, 2022-05-25:
The first firmware I installed gave me some weird results. 3 other floppy drive letters appeared.
[image]
FYI, I only have 1 drive. The drive is my internal Gotek drive.

The second firmware, gave me exactly the same results as firmware 3.28.
I have to reset my Amiga 600 and run the software selector. Then I choose an ADF HD image, press F7 and reboot.
Then it boots into my desktop.

[image]

I hope this information helps??

Regards,
Alexander.

keirf, 2022-05-25:
So in summary, the second firmware fixed the issues you have seen in 3.29-3.31, and works just as well for you as 3.28?

Tallguy297, 2022-05-26:
I tried the new firmware this morning. Same results as the previous firmware. It performs the same as firmware 3.28.

Thankyou,
Alexander.

keirf, 2022-05-26:
Thanks, that is good news! Let's try making the changes even smaller. Please give this one a quick go:
https://github.com/keirf/flashfloppy/suites/6666240569/artifacts/252332942

Tallguy297, 2022-05-26:
Same results again as firmware 3.28.
Boots from ADF DD no problem.
Only boots from ADF HD using software selector first.

Thank you,
Alexander.

keirf, 2022-05-26:
Has any older firmware version ever worked better than that? 

If you switch between DD and HD disks within Workbench, do they get recognised correctly each time?

Tallguy297, 2022-05-26:
I have no problem switching from DD to HD ADF images within workbench. Works perfectly, no read/write errors.
Yes older firmware versions have worked back to version 3.13, when I first starting using a Gotek drive.
However I have to specify `interface=amiga` in the FF.CFG file for this to work.

Regards,
Alexander.

keirf, 2022-05-26:
When you say older firmware back to 3.13 has worked: Has it worked to boot HD ADF without needing to run the selector? And if so, do you know which version broke that?

This would specifically mean: Does Kickstart 3.2.1 work with for example 3.13 firmware to boot HD ADF directly, with no selector?

Tallguy297, 2022-05-26:
That's a good question. I haven't thought to go back to previous firmwares to try it out.
However I do recall that it never worked rebooting straight into a ADF HD image without using the software selector.
I did try the other way, by removing the AUTOBOOT.HFE & the CFG file and booting that way, but it didn't work.

Regards,
Alexander.

keirf, 2022-05-26:
Ah yes, that's all correct. Like I say: There will be new Goteks available soon which could support direct boot correctly without needing hardware modding. It will depend how important that ability is for you.

Back to the immediate problem of a clean fix to get behaviour at least as good as 3.28... I'll have a bit more of a think!

keirf, 2022-05-26:
Okay here are two firmwares. Number 1 is a proposed fix, fingers crossed it works (but it might not!). Number 2 is a "control" experiment which I think probably should not work. If (1) works and (2) doesn't, we're done!
Number 1: https://github.com/keirf/flashfloppy/suites/6669164803/artifacts/252513402
Number 2: https://github.com/keirf/flashfloppy/suites/6669178307/artifacts/252514492

Tallguy297, 2022-05-27:
I tried both firmwares.
FIrmware No. 1 did not work. I could not boot from either ADF DD or HD images, which includes the software selector.
On the workbench I got the message to insert device DF0:.

Firmware No. 2 did work to a certain degree. I was able to boot from both ADF DD and HD images including the software selector. However if I chose to boot from an ADF HD image using the software selector then this did not work. The chosen image on the Gotek display stuck at track zero. When I booted into workbench I received the error to insert device DF0:. Changing images on the selector did not work either in workbench.
When booting into workbench and then inserting the USB thumb drive I was able to see both ADF DD & HD images, but got disk validation errors when trying to read ADF HD images.

Regards,
Alexander.

keirf, 2022-05-27:
Thanks for the report. I must have messed up firmware no 1 somehow. I'll have a look!

keirf, 2022-05-27:
Did firmware no 2 behave any differently than 3.31? That would be strange as the code is identical in both!

Tallguy297, 2022-05-27:
I can't think of anything else but what I have already stated. Sorry.

Regards,
Alex.

keirf, 2022-05-27:
Okay, I made some further effort to really optimise the interrupt handler. It is specifically compiled with `-Ofast`, plus some other special hacks to make it more concise. Please can you try the following, which we will call test firmware 3:
Number 3: https://github.com/keirf/flashfloppy/suites/6687609747/artifacts/253723624

keirf, 2022-05-27:
Yes I see this problem with an HD image selected at boot time too (your issue number 3). I think this behaviour is the same for older firmwares including 3.28, right? That is, it's not a regression? I *think* it will need better HD ID support, involving the MOTOR signal, to resolve it.

If you agree with the above, I think we're done and I can sort out a 3.32 release!

Tallguy297, 2022-05-27:
Sure no problem.
Thanks for all your help.

Regards,
Alexander.

keirf, 2022-05-28:
As you will see I released 3.32 with the fix included. However I now realise I probably optimise one routine too aggressively and a large function gets inlined into it. I think it will probably work okay with optimisation disabled on that one routine. Could you possibly try this firmware and compare it to 3.32? You can go back to 3.32 afterwards. Thanks!

https://github.com/keirf/flashfloppy/suites/6697857602/artifacts/254443551

Tallguy297, 2022-05-29:
I tried your new firmware 3.32 with the fix.
I tried the same tests as before booting in ADF DD & HD etc.
I didn't notice any difference.

Thank you,
Alexander.

keirf, 2022-05-29:
Thank you so much for testing. I will save myself 108 bytes SRAM in the next release.

closed 2022-05-29 by keirf