keirf/flashfloppy #651: Possible issue with dragon computer DSK files

johannfenech, 2022-05-27:
Hi Guys,

New user here, just got my hardware and trying to get it to run on a Dragon 64 computer, using a coco FD-502 running on a Superdos E7T Eprom and a "twisted" floppy cable. I also tried a straight one with the same result.

Gotek Firmware is latest FlashFloppy 3.31 (Originally the unit came with 3.30, but behaviour was exactly the same)

Gotek Hardware is **SFRKC30.AT2**

Jumper S0 is closed, and I have a piezo speaker connected. I also added a rotary encoder to the unit, which works great.  

I managed to mount VDK and DSK files by formatting a USB stick to FAT 32, copied  the DSK / VDK files there (in various sub directories) and finally added the attached FF.CFG file to the root directory (only) not in sub dirs.

Command like DIR and LOAD work fine even though most programs and games seem to crash after they start, unsure if this is due to corrupted VDK's / DSK's or I am missing something maybe.

Also any attempt to boot to OS9, starts the boot process, and a screen saying NITROS9 BOOT appears, but Gotek / Disk activity stops dead as soon as this initial screen appears. I know this is a known good VDK file, because I am able to boot to OS9 using a COCO-SDC which is basically  a floppy emulator using SD Cards, with the same VDK file.  

I am attaching my FF.CFG file for reference, because my suspicion is that I have something wrong or missing in it. Or maybe I require some other specifig DSK cfg file ?

Any pointers in any direction would be greatly appreciated 

[FF.CFG.txt](https://github.com/keirf/flashfloppy/files/8784579/FF.CFG.txt)

keirf, 2022-05-27:
I think you are set up correctly. Possibly others have had problems in the past eg:
https://forum.vcfed.org/index.php?threads/model-4-external-disk-drive-hxc.74766/
(apparently unresolved?)

johann-finixio, 2022-05-29:
Hi Keirf,

First of all thank you very much for finding the time to answer. It seems you are right. I will dig a bit deeper, and if I find anything useful, will revert.

Thank you very very much ... apart from the OS9 issue, the rest of the files work perfectly !

keirf, 2022-05-29:
Also update to 3.32. Probably makes no difference but always worth testing with latest (as of yesterday!)

johann-finixio, 2022-05-29:
Updated to 3.32, same result. But thank you very much for the tip.

J

prime6809, 2022-06-04:
If you are trying to boot stock Dragon Nitros9  (or Dragon OS-9), it will be expecting the Dragon Dos controller hardware,
so will not be compatible with the hardware in the CoCo controller. I think I have had that combination booting but you might need to roll your own boot disk for it to work.

johann-finixio, 2022-06-04:
Hi Phill,

Thank you for your reply. I think that is precisely the issue. In the mean time, I am fabricating a pcb based on [this](https://hardware.speccy.org/temp/Dragon-1.html) project. I have also ordered a couple of WD2797A-PL off ebay. Hopefully this will sort the driver issue, since this seems to be much closer to the Dragon's native controller. I have already prepared a 27C256 eprom with four different Dragon Dos firmwares (1.0, 1.2, 4.0, and 5.0) Hopefully the whole thing will work, and will get OS9 to Boot :).

With regards to rolling out my own boot disk, I am familiar with assembly languages, but I would not know where to start to be honest.

J

johannfenech, 2022-06-04:
Hi Phill,

Thank you for your reply. I think that is precisely the issue. In the mean time, I am fabricating a pcb based on [this](https://hardware.speccy.org/temp/Dragon-1.html) project. I have also ordered a couple of WD2797A-PL off ebay. Hopefully this will sort the driver issue, since this seems to be much closer to the Dragon's native controller. I have already prepared a 27C256 eprom with four different Dragon Dos firmwares (1.0, 1.2, 4.0, and 5.0) Hopefully the whole thing will work, and will get OS9 to Boot :).

With regards to rolling out my own boot disk, I am familiar with assembly languages, but I would not know where to start to be honest.

J

johannfenech, 2022-06-04:
> Also update to 3.32. Probably makes no difference but always worth testing with latest (as of yesterday!)

Tried version 4.x but with same results unfortunately

prime6809, 2022-06-04:
That controller should work perfectly once correctly aligned. That scan I did of the of the NDUG Dragondos schematic seems to get everywhere tho :) :)

Cheers.

Phill.

johannfenech, 2022-06-04:
Thanks a lot phil. Can wait for the WD's to arrive so I can put it together. Re - alignment anything I should look out for since I did not see anything about alignment though I saw it has a preset on the schematic / board

johannfenech, 2022-06-05:
Never mind, on the original schematic found what you are talking about TP's 1-5. Thank You ..

closed 2022-06-22 by keirf

johannfenech, 2023-06-19:
Just for the sake of completeness, I have managed to boot to OS9 using FF without any issues. Everything works perfectly. I am currently runninf FF 3.39. The issue was indeed the "wrong" disk controller. I have build my own Dragon Dos controller, and have tried several DDOS roms from 1.0 to 4.1. All work flawlessly. If it helps anyone, my ff.cfg is attached 
[FF.CFG.zip](https://github.com/keirf/flashfloppy/files/11788044/FF.CFG.zip)