keirf/flashfloppy #652: 8x16 font not working with 3.31

DL2THL, 2022-05-27:
In FF.CFG, I have configured the 6x13 font and "0,1" as the visible lines on my 128*32 OLED display, which worke. I could see the image name in the 1st row and trk/sec info in the 2nd row.
After switching to the 8x16 fon, I could only see the 1st row, the 2nd remained blank, although the 1st row used the larger font.
Having read about font issues that should be fixed with 3.31, I just did a FW upgrade to 3.31 without any luck, actually it got worse: Now I'm stuck with still only 1 line of text, but this time back at 6x13, no matter what I configured in FF.CFG.
Other than that, folks, this little gem is truly a masterpiece of art, keep going!! Thanks a lot!

keirf, 2022-05-27:
Yeah it's 5x9-ish within a 6x13 box per character, for kerning. I broke the 8x16 font in 3.31, I fixed it in the tree now, so it will be working again in 3.32.

Here is the test firmware which fixes the font issue (need to be logged in to GitHub for the link to work):
https://github.com/keirf/flashfloppy/suites/6667624640/artifacts/252417359

It sounds like there may still be some remaining issue with 8x16 font plus custom display-order? Let me know if this is still a problem with the above firmware.

keirf, 2022-05-27:
Just tested the font and display-order on the above firmware, and it does work for me.

DL2THL, 2022-05-27:
Nope, just flashed my Gotek, but the font size remains the same. It does display that ea44c1d on startup, though, so the flashing succeeded.

keirf, 2022-05-27:
Hmm
[image]

keirf, 2022-05-27:
It definitely works, so double check `oled-font=8x16` in your FF.CFG on your USB drive.

keirf, 2022-05-27:
Oh, and double check `display-order=0,1` -- no spaces are allowed in the comma-separated list of rows! The only place you can have spaces is around the `=` sign.

keirf, 2022-05-27:
Here is my display with an image mounted with above config:
[image]

DL2THL, 2022-05-27:
yep. I did revert to the old setting of 6x13 before, so that's why I again saw the small font. And the 0,1 didn't get recognized because for some reason I put it between single quotes, maybe I did a cut&paste from the comments. Anyways, it now works here as well, and rest assured that i never doubted that the fault was on my side, just didn't know where! Thanks again!

DL2THL, 2022-05-27:
One thought... would it make sense to display the line number of a line that cannot be parsed correctly? That way the user would immediately get visual feedback that his FF.CFG isn't syntactically correct.

keirf, 2022-05-27:
It's a thought but would take some work to be honest. Anyway, glad it's working!

closed 2022-05-27 by keirf