keirf/flashfloppy #653: OLED only displaying static

99z, 2022-06-01:
I'm unsure if this is a FlashFloppy issue but figured I'd ask here.

I purchased this for my 500: https://amigastore.eu/en/676-internal-usb-floppy-emulator-gotek-for-amiga-500-600-and-1200.html

Functionally, it works fine. I can boot into the FlashFloppy Selector, load adf files and play. But the OLED always displays static, as shown below:

[image]

Neither of the buttons on the side have any effect.

keirf, 2022-06-02:
Some OLEDs can be problematic. With latest FlashFloppy you could try `display-type=oled-128×32-slow` in FF.CFG to talk more slowly to the OLED.

closed 2022-06-09 by keirf

keirf, 2022-06-09:
Then I would say your OLED is bad and you should seek a replacement.