keirf/flashfloppy #656: booting floppies on x86 Pentium 1 computer

solvethispuzzle, 2022-06-09:
I recently installed flashfloppy on a FDD-UDD U144K using the ST-LINK V2.
I am using a Pentium 1 computer running windows 95 with what I think is the normal 34 pin floppy connector.
The floppy emu works fine when opened in windows or DOS on the machine but when I try to boot from the drive I get a non system disk error.  Currently the jumper is set to J1 but I have tried J0 and J2.
The flash drive is formatted fat32 with only one DOS 6.22 bootable .img file.
It could also be an issue with my gotek drive itself as I did not try to boot from it before flashing it.
Any help is appreciated.

solvethispuzzle, 2022-06-09:
My board does not have anything labeled JC or S1 just J0-J6 and 07,46-49
I currently just have a single jumper on J1.

keirf, 2022-06-09:
Oh I think it's one of the rare older Goteks then. I dont know so much about jumpering those.

keirf, 2022-06-09:
You want interface=ibmpc in FF.CFG. The 0_ display means the drive is accessing cylinder 0, lower head.

solvethispuzzle, 2022-06-09:
The booting issue is resolved
Thanks
For future reference the only changes I have made to FF.CFG is changing interface to ibmpc and I have my jumper on J1.

closed 2022-06-09 by solvethispuzzle