keirf/flashfloppy #662: FF & all versions of OS/2 1.x, Microsoft or IBM

csulaguy, 2022-07-02:
I wasn't sure if this problem was Gotek related or my controller. I'm using Gotek on both an IBM 5162 (XT 286) and 5170 (AT). Both give me the error "A disk read error occurred. Insert a system diskette and restart the system."

I can start Xenix and DOS okay, but not OS/2. I can also view the contents of said OS/2 disks in various versions of DOS okay.

I googled this issue, which led me to the issues page here on github. I upgraded to the latest 4.4a (I was previously on 3.29), but nothing's changed. Not sure if there are any flags I could try in ff.cfg, or if there's some other yet undiscovered issue?

keirf, 2022-07-02:
Use latest stable release, 3.33. Are the os2 disks in IMG format? I wonder if os2 track layout had any sector skew or interleave. It would be interesting to see dumps of original os2 disks.

csulaguy, 2022-07-02:
That was a fast response! I was going to also say that this is my ff.cfg as well:
`interface = ibmpc 
host = pc-dos
pin02 = auto
pin34 = auto
write-protect = no
side-select-glitch-filter = 0
write-drain = instant
index-suppression = yes
head-settle-ms = 12
motor-delay = ignore
chgrst = step
ejected-on-startup = no
image-on-startup = last
step-volume = 20
display-on-activity = yes
display-order = default
display-type=oled-128x32
track-change=realtime`

csulaguy, 2022-07-02:
> Use latest stable release, 3.33. Are the os2 disks in IMG format? I wonder if os2 track layout had any sector skew or interleave. It would be interesting to see dumps of original os2 disks.

Unfortunately, these are images obtained from both archive.org and winworldpc.com. I've done some dumps myself back in the 90s with IBM's savedskf.exe utility, but I don't think anyone was really thinking about flux readers back then.

The disk images are in IMG or IMA format. I can boot them up (albeit with an OS/2 trap error) on Oracle VirtualBox, which tells me the images work. I can also view all the views via 7zip, and it shows appropriate boot code marked as hidden and/or system. I'll downgrade to 3.33 and see if there's any changes.

keirf, 2022-07-02:
Latest 3.33 firmware will tell you if you have an stm32 chip. Either it will report 32k or 64k ram on the oled when no stick is inserted. 

You might want to try host=unspecified in FF.CFG and leave geometry detection based on image size alone.

csulaguy, 2022-07-02:
Mine has 64KB. Just downgraded both the bootloader and firmware to 3.33.

Going to try host=unspecified and see if that changes anything.

csulaguy, 2022-07-02:
Still a no go. Makes it as far as T00.1 as before and gives the same error.

keirf, 2022-07-02:
What sizes are the OS2 disk images?

csulaguy, 2022-07-02:
1440KB. Or to be exact, from a command prompt, 1,474,560 bytes.

keirf, 2022-07-02:
If they're all 1m44 then they should be handled correctly regardless of `host=pc-dos` or `host=unspecified`.

There was a much older OS/2 issue ticket that sort of petered out.... #26 

But I haven't really heard anything since.

csulaguy, 2022-07-02:
I read that and figured you closed the ticket, since there was no further activity. I almost wondered if it was a problem with my controller, so I think I even went as far as trying 2 MFM controllers and a SCSI controller.

csulaguy, 2022-07-02:
I'll do some more investigation on my own part and report back.

keirf, 2022-07-02:
Perhaps you can try loaddskf some image files? To HFE image format could be interesting.

csulaguy, 2022-07-02:
This is embarrassing... I'm able to replicate the same problems on a physical floppy drive. The same images boot just fine from virtual machines. It's getting late here, since I'm about 6-7 hours behind you, so I'll try over the weekend writing this images from loaddskf directly to an image file and/or floppy from other vintage equipment and report back.

csulaguy, 2022-07-02:
Made some more progress. Got out a Pentium 133 motherboard I use for testing, and the Gotek worked with the OS/2 images on there, but not on the MFM controller or an Adaptec 1542CF SCSI controller on the 286s.  I think this might be beyond the scope of your help at this point, but any suggestions would be great. I'm not sure if it's a controller problem, or the way the computer is handling the controller.

keirf, 2022-07-02:
I'm not very familiar with ancient PC floppy controllers but could OS2 simply be incompatible with your controller cards? DOS would almost certainly access the drive through the BIOS, but OS/2 would probably have its own driver and may work only with a standard motherboard FDC which doesn't sit behind ISA, SCSI etc etc

closed 2022-07-03 by keirf