keirf/flashfloppy #663: New version of SFRKC30.AT2; IO issues

bangstk, 2022-07-04:
Hello,

I have SFRKC30.AT2, but unlike what is stated in the wiki, it DOES have a JC header, and a corresponding trace on the board. It seems there is a new revision of SFRKC30.AT2 despite the same name.

[image]

Incidentally, I am having IO pin issues. Using the 3.33 IO test firmware:
WGATE is stuck reading as LOW even if 5V is applied
TRK00 is always outputting 5V and does not toggle LOW with the other outputs

Is it possible the board rev switched some of these pins around? Is there anything I should test to find out? Or is it more likely I have board/MCU damage somehow?

keirf, 2022-07-04:
Highly unlikely the board is bad and the JC addition is expected. I can fix io test for this board for you to try.

keirf, 2022-07-04:
Is it usart tx (pa9) that jc is connected to?

bangstk, 2022-07-04:
Having trouble connecting to the Artery site to get the datasheet; but JC here is connected to the 3rd pin from the left on the bottom side of the MCU (bottom being LED side of the board)

keirf, 2022-07-04:
Yes that is pin PA9. That was the pin I agreed with factory and JC should just work in that case.

keirf, 2022-07-04:
I have fixed the io_test, so please try the update file from the following automated build. Just click on the artifact link at the bottom of the page (you will need to be logged into github):
https://github.com/keirf/flashfloppy/actions/runs/2608421537
Let me know how it goes!

keirf, 2022-07-04:
I'd also like one of these new .AT2 boards for my test collection, if anyone has ordered multiple and has a spare!

RetroClinic, 2022-07-04:
I've looked at my stock, and without taking them apart, my .AT2 don't have the letters JC printed adjacent to the 3rd header from the right, so I presume that's the older version? The new ones I got last week are .AT3 with the larger chip.

keirf, 2022-07-04:
> I've looked at my stock, and without taking them apart, my .AT2 don't have the letters JC printed adjacent to the 3rd header from the right, so I presume that's the older version? The new ones I got last week are .AT3 with the larger chip.

That would be right. The new version has JC printed. Thanks for checking!

keirf, 2022-07-04:
See fixed io-test in v3.34. Please open a new issue if there are further problems.

closed 2022-07-04 by keirf