keirf/flashfloppy #664: Modern PC (i.e. Windows 10/11) support?

NeonWabbit, 2022-07-06:
I connected a GOTEK with FlashFloppy to my Windows 11 laptop via a 34-pin floppy to USB cable and surprisingly it showed up as an A: drive!

However, attempting to open the drive via the File Explorer with a 720K floppy IMG selected on the device understandably provided nothing, so next I tried selecting a 1.44MB IMG file and it actually _did_ provide something... this unusable garbage data:
[image]

So now that we know the GOTEK and FlashFloppy can communicate _somewhat_ with Windows 11, basically my question is: is there _any_ way of configuring FlashFloppy so that it actually works with modern PCs and doesn't provide garbage data? Such as if a delay was added, or if specific settings or jumpers need to be set? Not to specifically work with Windows 10/11, but perhaps as a way to install MS-DOS and early versions of Windows on modern hardware that don't have an OS installed, and for preservation/history purposes.

Thanks in advance for any suggestions!

keirf, 2022-07-06:
Is the IMG file known good? Have you jumpered S1 and JC at the rear of the Gotek?

NeonWabbit, 2022-07-06:
Yep, the IMG files work fine when loaded into virtualization software like VirtualBox or VMware. The jumpers included with the Gotek unit I have were not jumpered at S1 and JC, so I have tried that, but it is still reproducing the same results. I am unsure about the adapter itself, but it's one of those cheap ones you get from Amazon upon googling for such a thing. An image of the back of the device as it currently stands now is below:
[image]

keirf, 2022-07-26:
I don't really have more to add to this. Unless there's more specific advice I can give you, let's close this down.

NeonWabbit, 2022-07-27:
Right, I have recently pulled an old floppy drive out of a vintage PC and confirmed that the 34-pin to USB adapter works... but still not on the GOTEK with FlashFloppy, so for now I shall continue using physical floppy disks for this thing I'm working on, and leave the FlashFloppy GOTEK for a future vintage computer project, like an AMIGA. Thanks!

closed 2022-07-27 by NeonWabbit