keirf/flashfloppy #668: autoswap not work | Flashback

StevenHachel, 2022-07-23:
Hey,

I'm using a Gotek and trying to start flashback.
I keep getting the error "flash_autswap failed return code 10".

Hardware:
Amiga 1200 REV 1.D1 / Commodore Electronics
Gotek - GOTEKsystem | SFRKC30.AT2
Vampires ICEDRAKE (A1200)
AmigaOS 3.2.1

keirf, 2022-07-23:
Have you had any success with other Autoswap games?

How do you install the ADFs onto your USB drive, do you simply copy them all in, names unchanged?

StevenHachel, 2022-07-23:
hey,

one game actually works and that would be "Mortal Combat".
The following games do not get beyond the shell:
- Beneath a Steel Sky
- Goblins
- Indiana Jones and the Fate of Atlantis

I copied the directories directly from the AutoSwap directory one to one.

Steven

keirf, 2022-07-23:
I'll have a look. I fear incompatibility with kickstart 3.2.1.

JetA2, 2024-01-13:
I have the same issue with Atlantis. 
A500, KS 1.3
Gotek AT32F435, FF 3.38

keirf, 2024-01-14:
Really should work with KS 1.3 on an A500. You just get "failed return code 10"? How much RAM has your A500?

JetA2, 2024-01-14:
Yes, just "atlantis_autoswap failed return code 10".
I have 1Mb with the trapdoor expansion.

keirf, 2024-01-14:
Should work. I can repro that environment at some point and see what happens. Have you tried update to 3.42 firmware? Unlikely to fix this but always worth a try with latest.

JetA2, 2024-01-14:
I'm using my Gotek without display with a Pi Zero W permanently connected as a "smart" USB stick. I need to get me a display before I dare update so I can get some feedback during the update process.

Can the fact that I'm using the ```image-on-startup = static``` in FF.CFG mess with AutoSwap?

Also, I forgot to say thanks for FlashFloppy, it's awesome!

keirf, 2024-01-14:
No it really should just work iirc. I haven't actually developed or tested Autoswap in a long time. Therefore there is some chance it's completely broken on AT32F435, for example. I need to dust off the test setup and give it a try myself.