keirf/flashfloppy #672: Verify error with Roland S10

Pez67, 2022-07-27:
Hello. I am new here.. trying to get my Gotek FF drive to work with my Roland S10. I am running the latest firmware 3.34 and when I try to save to a blank.qd or any file for that matter I receive a "verify error" on the display of the Roland signaling there is something wrong with the qd.  Could someone share with me what the FF.cfg and Img.cfg files ought to look like for the S10, and/or explains the user process to save to the USB emulator as perhaps I am not dialing up appropriate slots for saving content?  Note that i am able to load files from the qd files on the USB. I only get errors when I try to save to the usb. Thank you.

keirf, 2022-07-27:
What's the Gotek board type (written on the rear of the PCB) and what are the jumper settings?

Pez67, 2022-07-27:
Hi,
I have attached a photo of the jumper connections. The board type is
SFRKC30.AT2

Thank you for your help
Chris

keirf, 2022-07-27:
Github drops attachments from email.

keirf, 2022-07-27:
You probably want a jumper at JC, and your Gotek probably has no JC jumper position.

What you will need is `interface = ibmpc` in your FF.CFG config file.

keirf, 2022-07-29:
Please reopen if further assistance is required.

closed 2022-07-29 by keirf