keirf/flashfloppy #676: SFRKC30.AT3 - something corrupts display, but display isn't broken

shea1337, 2022-08-02:
Here is an unusual one that I've only been able to reproduce with the new SFRKC30.AT3 PCB's so far. I found 4.6a and 3.31 Stable to be affected. (sorry I haven't tested a stable release since 3.31).

This has been happening with units that tested fine after flashing and construction with an OLED display, and were also fine when initially installed into a host device, and showed these symptoms upon a power-on but not necessarily the first power-on.

I at first thought my OLED was busted but it turned out the OLED was not the issue. I found the normal "Flashfloppy #.##" boot screen was displaying normally when removing the Gotek from my host device and powered it by a USB A-A cable.

The attached photo is a SFRKC30.AT3
 with 4.6a installed in a Roland S-760. Nothing crazy going on here. I booted it up, let the OS load in, let it run for a few minutes, and then power cycled and did the exact same thing 2 more times. It was fine for the 1st and 2nd boot up however the screen was immediately as pictured upon the 3rd power on.   It should be noted the USB stick was already inserted during power on when this happened. Each time this happened, I then disconnected the Gotek and powered it by the TTL programmer solely with no USB stick inserted and each time the "Flashfloppy #.##" boot screen appeared without any of the "snowy" display corruption which makes me wonder if something is getting corrupted on the USB stick itself or if it's a particularity of the new PCB or MCU that needs to be addressed

 FF.CFG was totally unmodified except interface=shugart was explicitly defined - everything else default.

As I previously mentioned, I was experiencing the same thing with this model PCB and FF 3.31 Stable as well.

I will be flashing a SFRKC30.AT3 with the latest stable+logging firmware to see if I can get any hints there and I will also be flashing a SFRKC30.AT2 with 4.6a to see if I can get the same thing to happen on that unit.

[image]

keirf, 2022-08-02:
Try the -slow modifier on display-type. That would be
```
display-type = oled-128x64-rotate-slow
```

shea1337, 2022-08-03:
> Try the -slow modifier on display-type. That would be
> 
> ```
> display-type = oled-128x64-rotate-slow
> ```

Thanks for this - it seems to mostly help, but not 100%.  I am on 4.6a but can test with the latest stable as well.

Since applying the -slow modifier, I was able to get stuck on the snowy screen once while shutting off immediately after a disk had loaded in, unfortunately I forgot to reload logging firmware on this one, so I do not have a log from when it went snowy.

To get it back to normal, I power cycled a few times quickly, it stayed snowy, then left it off for about 15 seconds and powered on again, and the screen was back to normal.

I have power cycled a few dozen times since then in different circumstances since then and haven't got it to happen again.

Let me know if I can check anything else

shea1337, 2022-08-03:
Of course I spent all day power cycling, make my post and then immediately get the snowy screen after power cycling again.

Leaving it off for ~1 minute and then powering it on without a USB stick inserted did nothing, but upon plugging in a USB stick and powering on again it was fine again

The log is attached.
[FFLOG.TXT](https://github.com/keirf/flashfloppy/files/9247928/FFLOG.TXT)

keirf, 2022-08-03:
So when it goes snowy you think it seems to remember that state for some time after power off?

keirf, 2022-08-03:
If you have a Gotek that seems immune, swap the oled (or gotek pcb) carefully tracking what goes where! See if the fault follows the pcb or the oled.

shea1337, 2022-08-03:
> So when it goes snowy you think it seems to remember that state for some time after power off?

It seems like it's kind of unpredictable as to what gets the OLED to initialize properly again. I tried swapping in different USB sticks with different FF.CFG's, tried getting into the FF Update Flash mode,  but for whatever reason, if I completely removed it from the host device for a short time, it seemed to go back to normal next time I plugged it in, as opposed to if I left it hooked up to the host device, where maybe some miniscule amount of charge was left over elsewhere in the circuit, it might remain snowy after even a minute or so of being powered off.

I will get a couple more AT3's set up with OLEDs and then see if any seem more tolerant than the others, thanks!

keirf, 2022-08-03:
My gut feeling is it's an OLED batch problem. You'll have to do some more testing to confirm or deny that though.

shea1337, 2022-08-03:
> My gut feeling is it's an OLED batch problem. You'll have to do some more testing to confirm or deny that though.

I have some idential OLEDs from a batch from a few years ago that were always fine, and I have some 128x64's. I will get a couple built and post in here once I get those tested, thanks!

shea1337, 2022-08-03:
I'm gonna say you are probably right. With the different batch OLEDs, even with -slow disabled I can't make it happen. 
The fact that -slow isn't needed with the other batch OLEDs indicates to me that it is in fact an issue with this batch.

Meanwhile, the questionable batch seems mostly reliable as long as I have -slow on. I have 4 AT3 units simultaneously testing right now and it does actually seem to be OLEDs from this batch.

I didn't notice -slow because I haven't checked a recent FF.CFG in a while, I must admit. I bought a few batches of 128x32 from different vendors hoping it's just isolated to this batch.

Many thanks again for your help on this one! I'll close it myself, and will keep notes on the new batches in case of any funny behavior.

closed 2022-08-03 by shea1337