keirf/flashfloppy #678: gd32f105 support

xdemax, 2022-08-10:
hello! is it possible to support gd32f105rct6? its looks like a complete replacement stm32f105x. FF firmware is loaded, but does not work.  possibly due to a different frequency?

https://gd32mcu.com/data/documents/datasheet/GD32F105xx_Datasheet_Rev1.6.pdf

alk0v, 2022-08-18:
I also made Gotek with GD32F105 but havent flash it yet.

keirf, 2022-08-18:
Not supported. I can sell you AT32F435 chips if that helps?

alk0v, 2022-08-20:
Is it pin-to-pin compatible with stm32f105 and supported by your great firmware?

keirf, 2022-08-20:
The latest FlashFloppy versions include a new build which targets AT32F435. It is indeed pin compatible with old STM32F105 boards (but not the old Artery At32F415 boards).

closed 2022-12-19 by keirf