keirf/flashfloppy #679: Add Yamaha QY700 to "supported devices"

discatte, 2022-08-11:
Hey! I just got a SFRKC30.AT4 (with oled/rotary) on FlashFloppy 3.32

Putting this here since its different from other Yamaha gear and caused me some confusion.

QY700 is confirmed working with:

JC jumper on
S1 jumper on

keirf, 2022-12-19:
Done, thanks!

closed 2022-12-19 by keirf

daftpunker, 2023-04-01:
Hi all

After installing my Gotek SFRKC30AT 3 3.35 – 32KB on the Yamaha QY700 jumper on S1 and JC I can't save or load, In attempts the qy700 remains planted in "Executing" and the gotek does not react
[image]

Even trying to load not the qy700 does not see anything to load even though I select them from the gotek.

I enclose the tested USB sticks
[Qy700 Trial1.zip](https://github.com/keirf/flashfloppy/files/11129371/Qy700.Trial1.zip)
[Trial2.zip](https://github.com/keirf/flashfloppy/files/11129422/Trial2.zip)

Obviously I also tried to do the factory reset but nothing to do.
Do you have a usb stick ready and working to attach to see if it's a problem with the ff.cfg file?

Thank you for your attention

discatte, 2023-04-10:
Here are the files I used. empty ff config except the `interface = jc` and qy demo disk.
Let me know if that works, else I can dig it out and try some testing.

[flashfloppy_qy700_example.zip](https://github.com/keirf/flashfloppy/files/11192953/flashfloppy_qy700_example.zip)

daftpunker, 2023-04-11:
Hi, thanks for trying but nothing has changed. Consider that being a SFRKC30AT 3 I proceeded to do a factory reset before proceeding to insert the usb stick (fat32 formatted) with the configuration you sent me. The demo to load is not displayed on the QY700.

daftpunker, 2023-04-11:
@discatte I also tried adding a BLANK to your config to be able to save the data but it doesn't change. It stays "Executing..."