keirf/flashfloppy #680: Gotek installed in Spectrum+3 not recognising any files

DAZZADOOBY, 2022-08-11:
I've recently installed a Gotek drive on my Spectrum +3. The display appears fine but it doesn't recognise any files on the USB sticks I've installed. I've tried different sticks with no joy. 
[image]

keirf, 2022-08-11:
Either your Gotek is broken, or your USB sticks are not formatted to FAT32. If you insert a FAT32 formatted USB stick I guarantee the Gotek display will change. Unless it's broken.

keirf, 2022-08-11:
Oh no! I tell you what are your usb sticks now recognised in a PC? Probably not. Your power connector is wrong way round and you are feeding your sticks 12v and blowing them up one by one. Stop that.

DAZZADOOBY, 2022-08-11:
Recognised absolutely fine. I bought the +3 cable kit so it's not a voltage issue.Sent from my Galaxy

keirf, 2022-08-11:
That's good! Then see my original reply.

closed 2022-08-11 by keirf