keirf/flashfloppy #685: Gotek SFRKC30.AT2 does not work with DISK CHANGE RESET on PA14

dl6lr, 2022-08-20:
Hi,

I have an old STM32 based Gotek that runs fine with DISK CHANGE RESET on PA14 (SWCLK on programming header) and FF 3.35. I flashed a "new" Gotek SFRKC30AT2 with AT32F415 in QFN32 package with FF 3.35, but the chgrst=pa14 does not work with this model.
I traced the connection of the header to run straight to pin 24 of the Artery chip (and to pin 13 of the 74HC04).

What is wrong?

Regards,
Bernd

keirf, 2022-08-20:
This model cannot support CHGRST as PA14 is used for floppy pin 2. The MCU doesn't have any spare pins to move CHGRST either. I should mention this in the wiki.

closed 2022-08-20 by keirf

dl6lr, 2022-08-20:
Ah, well, thanks. So thats why I tried to get any information about the differences in the discussions section.