keirf/flashfloppy #686: Feature request: Handle unusual track numbers in image

dl6lr, 2022-08-22:
Some old Hewlett Packard test instruments provide some copy protection scheme. They use IBM 34 format with 256 bytes per sector, but writing cylinder 78 with 0x96 to the track id. The FDC used is a WD2793. 

`
--------  Sector header: Cylinder 78  Head 1  Sector 0 ------------------------
0008b1d4:             96          |    .   | ID: Cylinder                      
0008b1d5:                01       |     .  | ID: Head                          
0008b1d6:                   11    |      . | ID: Sector                        
0008b1d7:                      01 |       .| ID: Sector size                   
`
So the drive is being stepped to cylinder 78, but the sector header is written with track id 0x96.
If there could be a configuration possibility in IMG.CFG to handle that case, that would be great. Suggestion:
tid=0-255|auto (auto*)

Furthermore they use sector ids that do not start with 1, but this is already handled with IMG.CFG id=x.
My current IMG.CFG that handles a lot of the odds (simplified, some cylinders/heads are unused, but it is simpler to waste that space in the image file):
`
[4952A]
cyls=80
heads=2
rpm=600
mode=mfm
rate=500
secs=16
bps=256
tracks=0-77
  id=1
tracks=78
  id=17
tracks=79
  id=97
`

keirf, 2022-08-22:
Why not use HFE images and convert to/from IMG offline when needed?

dl6lr, 2022-08-22:
Well, if this solves the problem it is worth a try. Have to investigate how to create this HFE format...

keirf, 2022-08-22:
Do you have existing images to be converted or is a blank hfe to be formatted by the device OK?

dl6lr, 2022-08-22:
Empty image for first tests would be OK. I assume I have to remove the IMG.CFG from the stick. Will test...

keirf, 2022-08-22:
You do not need to remove IMG.CFG. Attached is a suitable blank HFE. However it is HD rate which might be a problem for AT32F415 based Goteks. So it depends on your Gotek model.

[a.zip](https://github.com/keirf/flashfloppy/files/9396842/a.zip)

keirf, 2022-08-22:
By the way it was generated using the `mk_hfe.py` script shipped with FlashFloppy:
```
$  python3 ./scripts/mk_hfe.py --rate 500 --rpm 600 a.hfe
Geometry: 80 cylinders, 2 sides
500kbit/s @ 600RPM -> 100000 Encoded Bits
Data per HFE Track: 25008 bytes, 49 blocks
```

dl6lr, 2022-08-22:
OK, the HFE file works. I used one created with the tool from Hxc2001.com and formatted the blank image in the instrument. Never worked with HFEbefore. I had to patch all images that I converted as they had the wrong data rate.
Now I have to find a way to combine two HFE files (blank image with all strange formatted tracks) with an image of the lower cylinders only.

keirf, 2022-08-22:
Can be done in a Python script fairly easily. Got a pair of files to merge? Which tracks?

dl6lr, 2022-08-22:
I have empty formatted HFE file with 80 cylinders. And image files with lower 77 cylinders (0-76). So I have to append cylinders 77 to 79 from the empty file to the truncated image files.
With raw images I did this with dd. If you have a python script to do this, I can apply it myself to my images.
BTW: Do you have a wish list or something to donate to you?

keirf, 2022-08-22:
I'll need to rustle up a script. A couple of images to test with would be useful. Zip them up and attach them here.

My PayPal id is my github profile email address. Cheers!

dl6lr, 2022-08-22:
Empty_Disc_4952A.hfe is the formatted disc. All other hfe are the lower tracks only. I included the original IMD files for reference.

[tmp.zip](https://github.com/keirf/flashfloppy/files/9397402/tmp.zip)

keirf, 2022-08-22:
Great. I will find some time to rustle up a script. Might be tomorrow. Thanks for the donation!

keirf, 2022-08-23:
Attached are new merged HFE files and the script used:
```
python3 hfecat.py 18261A_IMD.hfe Empty_Disc_4952A.hfe new.hfe
```

[new.zip](https://github.com/keirf/flashfloppy/files/9400353/new.zip)

dl6lr, 2022-08-23:
Thank you very much. Working!

closed 2022-08-23 by dl6lr