keirf/flashfloppy #687: JC jumper not being read

snhirsch, 2022-09-05:
I just received a Gotek from Keir that contains PCB SFRKC30.AT4.35 and had modifications for encoder and LCD.  I saw mention that some newer boards lacked the JC jumper, but this unit had one.  Long story short, I expected that leaving the jumper open would provide a 'shugart' interface.  After an enormous amount of grief I found that the jumper was apparently not being read and that an explicit 'interface = shugart' entry was required in FF.CFG.  Question is: Am I misunderstanding something or is there a bug in the FF+ code?

keirf, 2022-09-06:
It's possible that there is a soldering fault on your pcb that means jc is permanently closed. One thing you can do is run the logfile firmware which writes a log to FFLOG.TXT. One thing this logs is the interface mode.

I ran a 435 Gotek on my Amiga board just yesterday and it worked fine with no config and no jc jumper.

keirf, 2022-09-27:
Reopen if further discussion required.

closed 2022-09-27 by keirf