keirf/flashfloppy #689: Error rAn 032 uEr 4 6A

Croccy22, 2022-09-11:
Have just fitted a Gotek to a Yamaha V50 Synth, flashed with FlashFloppy 4.6a

The synth uses a 26 pin cable so have used an adapter to convert it. The drive comes on and f-f is displayed (even if I insert a USB stick which works fine in other GOTEKS).

If I press the right button I get a error message rAn 032 uEr 4 6a. I've searched around but kind find this error or anything remotely similar to it?

Any ideas on what this error code is trying to tell me?

keirf, 2022-09-11:
That's not an error. It's reporting 32kb ram and version 4.6a. If the usb stick works in other goteks then probably the usb port on this gotek is bad? The display should at least change from F-F.

Croccy22, 2022-09-11:
Have just tried the drive in my Amiga and it works fine. So must be something to do with this Synth. I have found people saying they run a standard Gotek in it with no issues. So I'm thinking it either doesn't like FlashFloppy or the 26pin adapter isn't working or has the wrong pins.

keirf, 2022-09-11:
If F-F continues to be displayed perhaps voltage is wrong or something? It would be weird otherwise for host to influence whether the Gotek recognises a usb stick

Croccy22, 2022-09-11:
Yep, it's exactly this. Just measured the voltage going into the adapter and it's 5v, but the voltage on the power connector of the gotek was 3.3v. I guess these adapters drop the voltage down for compatibility with other systems. But they provide a floppy power connector on the converter board too, so I've linked them together and it's now working. Would be really nice if they provided instructions with these converter boards lol.

closed 2022-09-11 by keirf