keirf/flashfloppy #695: SFRC922D but use GD32F103

denjhang, 2022-10-08:
[image]
[image]

This version uses a black case and it looks like you have to avoid buying this type of gotek too.

keirf, 2022-10-08:
I could be wrong but this is probably an old model?

denjhang, 2022-10-10:
> I could be wrong but this is probably an old model?

This device doesn't look old, and at the moment I've found many gotek models that use stm32f103 or ch32f103, which might be some new variant. But in general, none of the xx32f103 can use the existing flashfloppy firmware.

keirf, 2022-10-10:
Stm32f103 doesn't work as a usb host. The gd32 pictured does, though.

denjhang, 2022-10-10:
> Stm32f103 doesn't work as a usb host. The ch32 pictured does, though.

So is it possible for flashfloppy to launch ch32f103 version?I can cooperate with the test firmware. Otherwise it looks like I wasted money (black gotek for 100RMB).

keirf, 2022-10-10:
I'd need to know there will be significant numbers of them to be worth supporting. And English documentation for the chip.

Looking newly made and being actually newly made are two different things. We don't know if this is old stock.

denjhang, 2022-10-10:
http://www.wch-ic.com/downloads/CH32F103DS0_PDF.html

denjhang, 2022-10-10:
[CH32F103DS0.PDF](https://github.com/keirf/flashfloppy/files/9743599/CH32F103DS0.PDF)

denjhang, 2022-10-10:
As far as I know, gotek using ch32f103 version is currently available on Xianyu (China's online flea market), Taobao, Pinduoduo and other websites. And these goteks are usually very cheap, and since the ch32 version can't use flashfloppy or HXC firmware, many people sell this kind of goteks at low prices.

keirf, 2022-10-10:
There's no way FlashFloppy is going to work in 20kB SRAM: 32kB (AT32F415) is bad enough, and I'm encouraging users to get AT32F435 if possible. I strongly doubt that HxC would port to 103 clones either, although I can't speak for that project of course.

keirf, 2022-10-10:
I have added a warning to the wiki (https://github.com/keirf/FlashFloppy/wiki/Gotek-Models#unsupported-models) -- are there any defining characteristics to warn about: eg are they always display-less and button-less? Model number? etc
Thanks!

denjhang, 2022-10-10:
This model needs to be avoided:
Black gotek, the motherboard model is SFRC922D.
The display is missing, there are no buttons, just a green LED.

denjhang, 2022-10-10:
Also, is it possible to adapt flashfloppy to ch32f103 by adding external SRAM?

keirf, 2022-10-10:
These low-end MCUs do not have an external memory interface.

I have updated the wiki further with identifying information. That's all I can do for this ticket. Thanks for the info!

closed 2022-10-10 by keirf