keirf/flashfloppy #711: Sharp X68000 support

tdaede, 2022-11-03:
The Sharp X68000 has mostly-normal Shugart drives, however they have an added electric eject circuit. Unfortunately, quite a lot of software depends on some of the behavior, including the bootloader. The extra pins are as follows:

OPTION SELECT: one per drive. This works like DRIVE SELECT but for the extra signals. It also clears FDDINT.
DISK IN: Asserted (low) when a disk is in the drive and OPTION SELECT is asserted.
FDDINT: Asserted when DISK IN changes, cleared by OPTION SELECT.

Other less important signals:
EJECT: ejects the drive when OPTION SELECT is asserted.
EJECT MASK: locks the drive.
LED: led.

I was able to get some level of functionality by mapping DISK IN to pin 2 and setting that "low", as some software asserts DRIVE SELECT and OPTION SELECT together. However, ideally I need one more input pin for OPTION SELECT, and one more output pin for FDDINT. I was thinking of maybe using MOTOR ON for OPTION SELECT (in the case of the artery hw), any thoughts on which pins are free and/or easiest to use for this?

tdaede, 2022-11-26:
Tying DRIVE SELECT and OPTION SELECT together, and connecting DISK IN to DCD and setting it low works in most cases.

However, it also requires the MOR jumper to be connected on the FlashFloppy. I don't understand why that's currently required, as motor-delay is set to ignore.

keirf, 2022-11-26:
Definitely jumpered MOR and not MO? That is weird as on motor-delay=ignore the MOR pin is not sampled in any way.

tdaede, 2022-12-05:
I made a dedicated PCB to handle all of the x68k-specific signals (except for disk presence and eject, which are mapped to dcd and mor respectively). I'll see if that increases the reliability at all.

tdaede, 2022-12-12:
My dedicated PCB works and handles almost all of the weird control signals on its own. There are only two tweaks I need still for full support:

- Mode for DCD pin (2) to indicate floppy status as a level (independent of disk select).
- Mode to map MOR pin to eject button to simulate motorized eject (entirely optional but cool)

keirf, 2022-12-12:
For DCD look at the `chg1` branch. This adds option of `pin02 = chg1` which places ungated Disk Change signal on pin 2. Perhaps this is all you need?

tdaede, 2023-04-13:
Sorry for the long delay, I had some hardware issues to sort out first. I tried building your branch, however the value of pin02 stays high regardless of insert/eject status when pin02 = chg1.

tdaede, 2023-04-13:
FWIW I have uploaded the extra hardware I am using to interface these signals here: https://github.com/tdaede/flux68k