keirf/flashfloppy #712: ZX Multiface 3 write on dsk

merlinkv, 2022-11-05:
Hello Keirf,

It is currently impossible to write snapshot files generated with the ZX Multiface 3 devices on +3DOS dsk files.

I have tried all possible configurations, but snapshots that are written to disk (dsk file) then cannot be played correctly.

However, if I write the snapshots to a real disk and then create an image of the disk. That image reads perfectly from the Gotek and the games load without problems.

Any idea?

keirf, 2022-11-05:
Dsk files are fixed format: flashfloppy cannot change sector sizes or identities. 

So why not write to one of the dsk files you created from a real multiface-written disk? Assuming you can overwrite existing contents in this way.

merlinkv, 2022-11-05:
If a make an exact image from a real MF3 written disk I can read it from Gotek.

But I cannot write snapshots to these images.

keirf, 2022-11-06:
Run logfile firmware and grab a copy of FFLOG.TXT after attempting a write. Instructions are on the wiki.

Or use a blank double density HFE image instead. This is less restricted on what can be written to it.

keirf, 2022-12-19:
Do you still wish to pursue this issue?