keirf/flashfloppy #713: latest firmware shows OLED screen rotated 180 degrees and text is "interlaced"

gfernval, 2022-11-07:
When I flash a Gotek SFRKC30.AT4.35 that has builtin OLED screen and rotary encoder from factory with 3.37 firmware, the text in the OLED screen appears rotated 180 degrees and the letters of the screen are not solid, are "interlaced"
See attached image
[image]

keirf, 2022-11-07:
You need following in FF.CFG
```
display-type = oled-128x64-rotate
```

This wasn't needed previously as your Gorek was flashed with a custom build which specified the above option as "factory default".

gfernval, 2022-11-07:
Thanks, now it works perfectly. Now the I turn off and turn on the PC, the Gotek shows the screen correct, the setting in FF.CFG is saved in the NVRAM of the AT32F435? I use Windows XP, do you have an Artery ISP programmer compatible with Windows XP? Latest version 2.0.06 is not compatible with Windows XP, I have had to use a Windows 10 machine and very ofter flashing "hangs" until I suceeded

keirf, 2022-11-07:
Yes FF.CFG options are cached in the Gotek. It is overridden by new options on FF.CFG or reset via Factory Reset option on main menu (press rotary encoder when no usb stick is inserted).

The only ArteryISP programmer is the one on the Artery website. It isn't surprising that XP isn't supported. Even Windows 7 is a bonus at this point.

closed 2022-11-07 by keirf

gfernval, 2022-11-07:
I use Artery ISP 1.5.47 (compatible with Windows XP) to flash AT32F415 Gotek drives, I already tried to modify AT32F435RGT7_1024K.ATmap (from 2.0.06) file to be compatible with 1.5.47 (does not detect the flash of the AT32F435)

keirf, 2022-11-07:
I would imagine all 435 Goteks are shipping with FlashFloppy and then you can update anyway by USB stick and no need for ArteryISP.