keirf/flashfloppy #716: [Enhancement] Single Button / No Display Mode

insane-rabenauge, 2022-11-20:
All this is only for the HxC Compat Mode with a file selector:
Is there any possiblity for a configureable single button mode - i.e. one button press = next image; long press= image 0 (selector)?
Upon selection the beeper should beep X times where X=image slot selected - like the old pre gotek hxc floppy emus
This would allow using Flashfloppy with the selector software + only a single button + speaker without needing any kind of display.

Thanks :)

keirf, 2022-12-19:
Is one button vs two a big deal? It's quite a change to the UI to have to support one button only.

insane-rabenauge, 2022-12-28:
Not really per se - I believe having the beeper beep X times for slot selected in image mode would already be an improvement if no display is connected.

keirf, 2022-12-28:
So this would be auditory confirmation at the moment an image is selected and mounted? So for example, in default configuration, and image is auto-selected after two seconds of no button activity. It would be after those two seconds that beeps would occur?

Also I would suggest long beeps to count in fives, followed by short beeps for 1-4.

I suppose this could be a new display-type = none or audio. Or it could be a completely separate option.

RetroClinic, 2022-12-28:
It would be good to have a two tone beep, like the USB insert or remove sound on a PC, whenever you insert or deselect an image. I have the GOTEKs I sell setup for manual insert, so the user has to press the encoder to insert, and press again (or select something else) to unselect. This could be turned on or off independent of any display. Just an extra audio confirmation of the image being inserted.

Thanks, Mark.

From: Keir Fraser 
Sent: Wednesday, December 28, 2022 3:16 PM
To: keirf/flashfloppy 
Cc: Subscribed 
Subject: Re: [keirf/flashfloppy] [Enhancement] Single Button / No Display Mode (Issue #716)

So this would be auditory confirmation at the moment an image is selected and mounted? So for example, in default configuration, and image is auto-selected after two seconds of no button activity. It would be after those two seconds that beeps would occur?

Also I would suggest long beeps to count in fives, followed by short beeps for 1-4.

I suppose this could be a new display-type = none or audio. Or it could be a completely separate option.

keirf, 2022-12-28:
What an excellent idea. I could work that in too. The select sound could either be boodle-bip (low to high three tone sequence) or count out the slot number as described previously. And eject would be biddle-boop (high to low three tone sequence).

None of these would probably work with the active buzzer some people use, but anyone using a buffered magnetic speaker or a piezo speaker it could work nicely.

insane-rabenauge, 2022-12-28:
With mounting beeps and after then beeping a number of times according to the drive image selected this would really be fine.
I mostly use the file selector software on an Atari ST to set up my floppy slots and then I simply switch between them.
Using Long beeps and Short beeps is a great idea - I'd use Long for multiples of 10 and Short for 0-9 though. 
(Personally I never have that many images in a list at the same time).

keirf, 2022-12-28:
On my ROM selector firmware (separate project) I have the short beeps going pretty damn fast. Counting nine of them might be tricky. I will see how it goes.

keirf, 2022-12-30:
I have some chimes and some volume tables. I just need to plumb it in and make it configurable.

keirf, 2022-12-30:
Here is a test firmware to try. To enable the notifications including slot-number, add something like the following to FF.CFG:
```
notify-volume=5,slotnr
```
Note no spaces! And adjust the number to taste (it's the volume, from 1 to 15; 0 is off).

https://nightly.link/keirf/flashfloppy/actions/artifacts/493450888.zip

Let me know how you get on, and any comments for improvements!

By the way, slot number is identified by a sequence of long (+5) and short (+1) beeps. Let me know if you think this needs tweaking at all.

keirf, 2022-12-30:
PS Also interested in what volume level works for you. 5 is about right on my speaker. I can tweak the 1-15 range (much above 5 is too loud on my speaker, but there could be quieter ones).

keirf, 2023-01-05:
I'm knocked out by the enthusiasm for this feature ;)

insane-rabenauge, 2023-01-05:
:)
Still away from any 16bit Machine for the next days - I'll update you ASAP

insane-rabenauge, 2023-01-13:
It beeps majestically!
Doesn't beep while selecting, only after inserting the image but I can live with that.
Beep speed is fine - but I believe beep count should be limited to 100 or until a button has been pressed.
(Inserting disk image 575 in direct mode turned out to be a bad idea ;) )
Thank you very much!

keirf, 2023-01-17:
Managing 575 images with no display is not a brilliant idea I suspect. I'm thinking more like.... 20 images would be a sensible maximum?

closed 2023-01-17 by keirf