keirf/flashfloppy #719: Motor pin has an effect on operation even if motor-delay = ignore

tdaede, 2022-11-26:
If the MOR jumper is not installed (resulting in the motor pin being high), there is a visible effect on the RDY pin resulting in read errors even if motor-delay = ignore.

keirf, 2022-11-26:
Can you characterise the effect?

But yes any effect on rdy when motor=ignore is a bug. Motor should be assumed always asserted in this case.

tdaede, 2022-11-26:
When booting, I can usually read tracks 0-1 or 0-2 fine but then it fails after that. With the MOR jumper attached it can read the whole disk. I haven't been able to figure out yet if that's due to RDY line changes or some other effect.

I did double check the grounding but that also seems OK.

keirf, 2022-11-26:
Perhaps this host wants pin 16 pulled up for some reason?

tdaede, 2022-12-12:
I rebuilt the interface as a dedicated PCB and these problems seem to have gone away. I am going to chalk it up to signal integrity issues.

closed 2022-12-12 by tdaede