keirf/flashfloppy #720: New SFRKC30.AT4.35 not working with Amstrad CPC6128/664/464

SerErris, 2022-12-02:
Hi,
I have got one of the new Goteks with Rotary and Display. FF itself was already on it and I could update to 3.37. 

However I could not get it to work correctly as a second disk drive on an Amstrad CPC 6128 nor on an Amstrad CPC464 with DDI-1,FD1 and Gotek as a second drive. 

Both scenarios are exactly the same.

If you start everything, you cannot read from physical drive A as soon as a disk is in the gotek. If you remove the USB key from it, Drive A (the physical 3") drive works. It also works if you eject the disc (not the stick). Same result, Disk A start working again. As soon as there is a disk mounted in the Gotek, no read access is possible anymore. 

You can hear the motor spin, you can hear the head moving, but it fails to read.

I tried a lot of firmwares (starting with 3.33 and up to the latest 4.6a the result persists.

I have removed the drive type jumper and put the drive select jumper to S1. So this drive will answer only to drive B calls. 

The exact same setup worked with 3.37 on an older SFRC922D and still works.

I am not sure what is different between those two hardware versions, that makes one work and one not. But I would appreciate if you could look into this. Let me know what else you would need and I will collect it and run further tests.

Regards
Ser

keirf, 2022-12-02:
What is the model number of the offending Gotek?

SerErris, 2022-12-02:
You mean SFRKC30.AT4.35?

keirf, 2022-12-02:
Yes, so that the new model with 435 chip and uses a different firmware build. Still it's unlikely that the Gotek should be conflicting with other drives on the bus. It's at least as likely that your 435 Gotek has a hardware fault -- you should check that soldering is okay on the underside of the PCB around the jumper block.

keirf, 2022-12-02:
I've done a few quick tests with a pair of Goteks on an Amiga, and the 435 Gotek doesn't seem to conflict with the other.

SerErris, 2022-12-02:
I have two 415s and both have the same issue. 

i can check the soldering, but would doubt, that both are bad …

keirf, 2022-12-02:
ITYM "I have two **435s**"?

SerErris, 2022-12-02:
I did some more tests.

My setup is the following:
I do have drive A the 3" drive and as drive B the Gotek.

When I cold start the system I can access A. Then switching to drive B and reading the disk catalog (CAT command). That works. Then switching back and then it never works again. I can only make it working by ejecting the disk from the Gotek. That does work - however as soon as the disc is back in, it stops working.

It looks to me like the gotek after getting selected via DS1 one time, it does not let go anymore and also answers commands. Is that possible? How could I verify this theory?

Strange is that it works and then stops working after switching one time to the gotek.

Again the exact same firmware for my older SFRC922D works fine.

SerErris, 2022-12-02:
> ITYM "I have two **435s**"?

You are right, sorry for the confusion.

keirf, 2022-12-02:
My new weirdness repro scenario does not need to happen immediately after power on. I can cycle the following:
1. DF0 (435) read test. Exit the read test while the track counter on DF0 reads Tx.0 for any x.
2. DF1 (415) read test. T0.0 will not read.
This only happens when DF0 is 435. Needing to be Tx.0 makes sense. Then the HEAD signal is not changed again until DF1 reads T0.1, and that triggers `rdata_stop()`.

And this would explain why the problem is worse on CPC. Most images are single sided. In that case HEAD signal is ignored (CPC probably isn't toggling it anyway) and that `rdata_stop()` in the HEAD-change ISR is never called. So the 435 Gotek pollutes the bus forever.

Nasty bug. I wonder how hard it will be to track down.

SerErris, 2022-12-02:
Is that something you can fix in firmware?

keirf, 2022-12-02:
Thanks for your further testing. I think our tests are in agreement. When the 435 has been on the bus, it doesn't give it up the RDATA line, unless something else happens to cause `rdata_stop()` to be called. Like a disk eject, stick eject, or head change.

Yes, it's a firmware bug. It will be a firmware fix.

SerErris, 2022-12-02:
Any idea why that does not happen on my old STM32 gotek?

keirf, 2022-12-02:
I don't have any firm idea on the cause yet. It could be a latent race condition in the firmware that exists on all Goteks. The 435 runs 4x faster than STM32 and 415 Goteks, so the timings can be quite different and trigger such a race.

I can trigger this bug every time now, so I suppose I will turn on serial-line logging and work out what state the 435 Gotek has got itself into.

SerErris, 2022-12-02:
> And this would explain why the problem is worse on CPC. Most images are single sided. 

That is true for the builtin AMSDOS. But there other DOSes like PARDOS or VDOS, that can work with DD drives (80trackes, 2 sides, 9 sectors each). And this is one of the configurations where my drive is supposed to be in (VDOS this time). The DOS is from 1986 and was very prominent in Germany, as you could have a 5.25" drive with massivly cheaper disks than the very expensive 3" disks. And 4 times the capacity without turning the disk :-)

SerErris, 2022-12-02:
Thanks for your support so far. Let me know if I can help with something.

keirf, 2022-12-02:
Okay, here is a fix for you to try. You will need to be logged in to GitHub for the link to work. Use the regular firmware (ignore the debug zip file).

https://github.com/keirf/flashfloppy/suites/9641490140/artifacts/460878775

keirf, 2022-12-03:
Will you have time this weekend to give it a try?

SerErris, 2022-12-03:
Tried it and can confirm: it works. Tried with CPM and with AMSDOS (Basic) which are separate things . Both working as expected

keirf, 2022-12-03:
Thanks for testing. I will do a release this weekend as this is a pretty nasty bug if you hit it.

closed 2022-12-03 by keirf