keirf/flashfloppy #721: Roland MT-200 only partially works

peterbuxton, 2022-12-04:
I am using a SFR1M44-U100K Gotek unit with FlashFloppy-3.37 (tested with two different MT-200s). The Roland host section of your Wiki mentions settings for MT-120 (which I think is similar), but they disable the MT-200. Results of my tests are variable and inconsistent. The main issue is some errors while reading images, but always errors when writing. I'm guessing this may be due to quirks of the Roland image format, since Gotek/FlashFloppy works fine in a PC as a floppy disk replacement. Do you have any information to make the MT-200 work, and would you be interested in specific data from my tests?

keirf, 2022-12-04:
I don't have much info on mt200 so I suppose it is fairly rare. However it seems it works with factory firmware so presumably uses 720k or 1.44m pc formatted disks. What sort of image files are you using?

peterbuxton, 2022-12-04:
Here is the general procedure:
•Format a 1.44m diskette in the PR-300 disk drive, create and save song (sequence) to it, check that it plays back. 
•Use a Linux PC to create and save an image file of the diskette.
•Copy the image to various USB drives formatted as FAT32 in Windows 10. 
•FlashFloppy displays and selects the partition image names properly.

Results:
•The Gotek must have jumpers M0, S0, and FB (otherwise the PR-300S displays a disk error).
•Image file has a song in it which plays correctly. 
•FF.CFG line "pin02 = high" (without quotes) causes the PR-300 to display error message: Disk NOT PR-300S READABLE. 
•Removing the line, the PR-300S will work again but often still has the error until doing FlashFloppy Factory Reset.
•If I try to format that working image using the PR-300S, I get the error: Disk Write Error Format Failed. 
•I get a similar message if I try to save the song: Disk Write Error Save Failed.
•In both cases above, FlashFloppy shows a flashing "w" indicating trying to write, before the error displays.
•Images with some other songs load into the PR-300 display, but when I try to play them, they stop after a few notes and show: Disk Read Error.

Other tests:  
•Write Linux image file back to another diskette, put floppy drive back on MT-200, and the disk plays and writes normally, so it doesn't seem that my Linux method changes the image file.

It seems to me that I must have something misconfigured in FlashFloppy since the PR-300S worked better with the original Gotek system.

peterbuxton, 2022-12-04:
I'm sorry to have dumped so much into the last reply, and also didn't mention at first about the MT-200 actually being a PR-300 (which is the same). There are some mentions of MT-200/PR-300 on the Internet but most people now seem to use them as a Standard MIDI File player for games. I have seen several mentions (plus experience myself) of how the Roland-formatted diskettes can be read by DOS but not vice-versa until the Roland reformats them (which for one, rewrites the OEM ID bytes 03h-0Ah of floppy image, and maybe more, shown below).

xxd -l 16 xp-format.img
eb3c 904d 5344 4f53 352e 3000 0201 0100  .<.MSDOS5.0.....

xxd -l 16 pr-300fix.img
eb3c 9052 4a41 312e 3030 4200 0201 0100  .<.RJA1.00B.....

keirf, 2022-12-04:
What is jumper FB? Have you tried a range of makes/models of usb drive? What is the model of gotek as written at rear of the pcb?

peterbuxton, 2022-12-04:
At the rear of the pcb is SFRC922D. I don't know what jumper FB is but it was mentioned as needed in [video](https://www.youtube.com/watch?v=ib1qC7yRIoI&t=901s). The MT-200 only shows a disk error without it. I have tried three flash drives of different makes and two microSD cards in a USB adapter. 
[image]

keirf, 2022-12-04:
That would very odd as it looks like JB. But that is a speaker output on FlashFloppy. It can't affect emulation and putting a jumper there grounds the speaker output and risks damaging the microcontroller (though that's unlikely the problem here).

keirf, 2022-12-04:
Assuming you have an 888 led display, if you press right button with no stick inserted it should report ram amount. Likely either 32 or 64 kilobytes. Interesting what it says.

peterbuxton, 2022-12-04:
Actually, I did a complete upgrade per your site and it now has an oled display and rotary encoder. I left jumper JA off because I think that is where the encoder switch is connected.

peterbuxton, 2022-12-04:
Here is the original floppy drive used in the MT-200, and its jumper settings, if that helps.
[image]

peterbuxton, 2022-12-05:
My OLED display shows 64KB. Also, just I discovered that jumper FB is NOT necessary. I know that I tried it before and it failed, but now it has no errors with only M0 and S0 present. 

I use an image of a blank floppy formatted in the MT-200 that I can successfully put certain songs in using "mount -o loop" in Linux and they will play from FlashFloppy. Other songs that did play when on floppy disks fail to do so from the flash image. However, no song files that do load can be written to flash by the MT-200 and no images will format. Both actions only give an error in the MT-200 display. Formatting a flash image fails after showing track 80, then 79. Song save shows saving for at least several seconds but ends in a failure too. Both actions show a "W" in the FF display while trying.

Maybe all of these issues are Roland's proprietary design and not worth pursuing: the unit works normally using the floppy disk drive. However if it matters to you for your product, I'm glad to contribute further tests and info at your request.

keirf, 2022-12-05:
I suggest no FF.CFG and make sure the Gotek is factory reset (which you have done before, it's an option off the main menu with no usb stick inserted). Then jumper at S0 only. MO shouldn't really be needed. This gets you a definite blank slate that matches the real drive's jumpers (MM which is motor related doesn't matter -- Gotek ignores motor and assumes motor is always on).

Another thing to try is an HFE image instead of IMG. Really IMG should work better, but HFE is a "rawer" format and could show up if the Roland doesn't like something about the IMG emulation. Here is a blank HFE image you could try formatting in the Roland. I suppose it will be quickly obvious if it doesn't work:
[1m44.zip](https://github.com/keirf/flashfloppy/files/10150864/1m44.zip)

A final thing to try would be to gather an FFLOG.TXT file: https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive

This is definitely a weird issue because Roland hardware which uses PC-formatted disks is usually some of the easiest to set up.

keirf, 2022-12-05:
Another one: If you *do* manage to format the HFE image file, please attach it here. It could be that the disk layout is a bit different than a conventional PC disk. Eg sector interleave or track skew. If you can format a raw HFE image, such things will be evident when I analyse the image file. And then I can help you reproduce the same using IMG files, if you want.

peterbuxton, 2022-12-05:
I am going through your previous post. I removed FF.CFG, did the factory reset, removed the M0 jumper, and inserted the flash drive. The MT-200 displays the first song of the image of the demo diskette with 10 songs. If I push Play, or try to go to a different song, the display then says Disk Read Error.

Next, I put just the HFE image on the drive, reset the Gotek, plugged in the drive and the MT-200 displays Improper Disk.

Now I'm just ready to flash the Gotek with the logging file. I'll write back as soon as I finish it.

peterbuxton, 2022-12-05:
I did the update using the logging file instructions and after it updates, the Gotek shows Error 34 (because there isn't an image, I presume). When I take the USB drive out, the display shows FlashFloppy 3.37 Log 64KB. However, there is not any FFLOG.TXT file on the drive, only IMAGE_A.CFG of 0 bytes. I must be doing the procedure wrong.

keirf, 2022-12-05:
Okay skip the logging firmware for now. I think track layout is a more promising avenue. The HFE image is entirely blank, as if you wiped a magnet over a real disk. So it is no surprise that the Roland doesn't like it. However is it nevertheless possible to attempt to format the disk from the Roland menus? The interesting things to establish are 1. Will it format and 2. If so what does the format look like physically.

keirf, 2022-12-05:
Yes 40 bucks sounds about right for my USA reseller. You may yet resolve this using the HFE image however. And also this track layout hypothesis could be a blind alley.

That said a Greaseweazle is a useful thing to have in general if you are still working with floppy disks.

peterbuxton, 2022-12-05:
I will order a Greaseweasel from the link on your Github site. I am interested in working more with floppies in the future. 

In this case, if the HFE is the only image on the floppy, the Roland thinks there isn't a disk. When I go to the format menu, it tells me to insert a new disk. Since it doesn't see one, it won't start the format and just sits there until I press cancel. So I put a DSKA0000.IMG file along with the other. I got it to go to the format screen on that image, then switched to the HFE when it asked for a new disk to format. I pressed format and it did the same as all my other tests: it showed Track 80, then 79, then Disk Write Error Format Failed. So it fundamentally seems not to be able to write.

keirf, 2022-12-05:
A final roll of the dice I will make you a fat formatted hfe image to try. I will attach it here a little later on.

peterbuxton, 2022-12-05:
Thank you! I will quit now for the night but I'll be back in the morning. I will also order a Greaseweasel right now to get it as soon as possible.

keirf, 2022-12-05:
Okay, here is a zip containing HFE and the IMG it was generated from. Worth giving formatting of that HFE image a go.
[1m44_formatted.zip](https://github.com/keirf/flashfloppy/files/10152447/1m44_formatted.zip)

Other things to try... You could try the v4.x experimental firmware. This has a very different strategy for I/O buffering. The latest v4.x version is v4.7a.

Also, going back to the logging firmware, what you would do is install that firmware, then attempt, say, a format operation, and then eject the image, and then pull the USB stick and harvest FFLOG.TXT.

keirf, 2022-12-05:
Another theory I have is that pin 2 is density select from the drive to the Roland host. In which case you might want to try the following possibilities in FF.CFG: `interface=ibmpc-hdout` and `interface=jppc-hdout`

Be very careful of spelling! Also do not specify `pin02=` or `pin34=` options and make sure they are not cached by performing factory reset after changing your FF.CFG config.

peterbuxton, 2022-12-05:
Wow, thanks for the wealth of info! I will immediately start going through your latest messages and report back with results.

peterbuxton, 2022-12-05:
I'm going through your latest messages, kind of in the order you sent them. Using v3.37, none of the three HFE images could be formatted by the Roland but again, it just showed Track 80, then 79, then Disk Write Error Format Failed.
Next, I updated to v4.7a and got the same results. I wanted to make a log file using v4.7a but even though FF_Gotek-logfile-4.7a.upd appeared to update the system, the display only showed v4.7a and not v4.7a **Log**. So I went back to v3.37 and got it to create 
[FFLOG.TXT](https://github.com/keirf/flashfloppy/files/10156224/FFLOG.TXT)
FFLOG.TXT.

keirf, 2022-12-05:
Yeah no logging on v4.x makes sense. I had to disable all debugging on the 105/415 build to fit in Flash!

The FFLOG.TXT you attach shows no write attempt whatsoever. I assume you did try a write or format before harvesting FFLOG.TXT? I think with an IMG file this probably points to the MT-200 thinking it is writing a DD disk, whereas FlashFloppy is expecting HD sectors, and therefore cannot parse any sectors out of the WDATA passed to it by the MT-200.

Which brings us on to trying `pin02=dens`. Or one of the other methods of getting density select on pin 2, outlines above. Even `pin02=low` would probably do, since you are dealing in HD images only. But `dens` is more polite if it works.

keirf, 2022-12-05:
The other possibility would be a hardware problem with the WDATA or WGATE signals on the Gotek. This seems a bit less likely however, especially if you have written successfully on this Gotek before (eg factory firmware).

peterbuxton, 2022-12-05:
You've given me **a lot** of different things to try and I want to do the most meaningful tests possible. Should I keep using v4.7a or go back to v3.37 (or v3.38 as I see you've now updated your latest release)? Also, are we finished with the three HFE images for now? I'm concerned that FFLOG.TXT didn't show any attempt to write, since the Roland always posts errors about writing and thinks it is attempting it (I was using DSKA0000.IMG, one of my regular images then, not any of the HFE images). 

Also, I tried using FF.CFG with | interface=ibmpc-hdout | pin02=auto | pin34=auto | and DSKA0000.IMG (which contains one song) the first time displayed the song, but errored when I tried to play it, but after that would not display it even if I did a Factory Reset. It seemed to corrupt the img file, I think. Do you want me to try interface=jppc-hdout next? I'm sorry to take so much of your time.

keirf, 2022-12-05:
We are done with v4.7a so stick with 3.37 or 3.38.

You can tell if the img gets corrupted because you could take a copy beforehand and then md5sum the files (or similar) to confirm the files are identical. If you didn't attempt a write then it is unlikely the image file got corrupted.

Attach your FF.CFG if you would like me to check it when I am back at keyboard. 

Perhaps try `pin02=low` in FF.CFG. With IMG file.

Finally you could try formatting that HFE file while running the logging firmware. This should see some attempt to write even if it's a failure.

I don't suppose you have a logic analyser or scope?

peterbuxton, 2022-12-05:
Yes, I do have an oscilloscope.

Still using v4.7a, I just tried interface=jppc-hdout with DSKA0000.IMG with a song. The Roland displayed the song and played it. I tried saving and copying the song and it appeared to make a copy to the image. Then I tried formatting the image and it went through all the tracks down to 0. Then I recorded a new one and saved it. I rebooted the Roland and it found the song on the image and played it.  I think this is the answer! Could I do anything to refine it?

keirf, 2022-12-05:
My guess is that the mt-200 needs ready signal on pin 34. In which case `jppc-hdout` is perfect and should work on 3.38 too.

If that works, I'll update wiki for mt-200 and mt-300 and e-96 too.

peterbuxton, 2022-12-05:
Using v3.38 I just verified that I can have three images on a Kingston 8GB USB drive, and the songs in all three images will play, and I can copy, save, and delete them and move them to a different image. I can format an image and it is then blank, and then I can save a new song to it. That tells me that everything is working with the disk functions. 

The only question I have now is that possibly I might have to do a Factory Reset sometimes if functions seem to stop working right. I don't know if that's the case now but it has been before. Do you have an explanation on what to do in that case?

peterbuxton, 2022-12-05:
Thank you too! I am also looking forward to working with the greaseweasel to understand more about floppy disks. Sounds like a great tool.

keirf, 2022-12-05:
When you get the weazle feel free to dump an mt-200 formatted disk to a raw image format such as SCP and send it to me. I can check if there's anything special about the sector layout which could optimise access speed if emulated by the Gotek. Find me on Facebook or Discussion section of the Greaseweazle github pages if you need help getting set up.

peterbuxton, 2022-12-05:
Will do! Thanks again. I will close this issue if I don't hear any more from you.

keirf, 2022-12-05:
I'll close the ticket after I have updated the wiki. Otherwise I can forget.

By the way, you ended up with jumper at S0 only, right?

peterbuxton, 2022-12-05:
That's correct: S0 only, and I don't know why it was failing for me before when I tried it without the other jumpers. Anyway, you explained how it should work and I was able to confirm it.

peterbuxton, 2022-12-05:
LOL

keirf, 2022-12-05:
Wiki is now updated: https://github.com/keirf/flashfloppy/wiki/Host-Platforms#roland

closed 2022-12-05 by keirf