keirf/flashfloppy #732: Flashfloppy vs Hxc documentation

orcutt989, 2022-12-18:
We need some good documentation in the Readme definitely since thats the first thing people see.  Something comparing the pros and cons of Flashfloppy vs Hxc. I just got a Gotek myself and I see that many people upgrade to Flashfloppy, but its not clear exactly why from a functional stsndpoint. I think something detailing pros and cons would be a huge help and

keirf, 2022-12-19:
Switching from hxc to flashfloppy is not common because hxc is a paid license which you lose if you overwrite the hxc firmware. FlashFloppy is overwhelmingly applied to factory fresh Goteks because for most users there's little to pick between FF and HxC and FF is free!

orcutt989, 2022-12-19:
Im confused then what comes on this GoTEK SFR1M44-U100 3.5 Inch 1.44MB USB SSD Floppy Drive Emulator Black https://a.co/d/1t9TA7O? If not Hxc then a comparison of whatever firmware comes on these vs FF.

keirf, 2022-12-19:
The factory firmware is not HxC. It is a fixed format firmware supporting only 1.44MB images assigned to 100 partitions on a USB drive formatted with special software. It's pretty poor.

closed 2022-12-19 by keirf