keirf/flashfloppy #733: Apple II and/or classic Macintosh support

greenonline, 2022-12-24:
I've searched in both the open and closed issues for "Apple", "Mac" and "Macintosh" and not found anyting obvious (so, apologies if this has been asked before).

This is a *feature request* for the ability to connect the drive to an Apple II or a classic vintage Macintosh.

I realise that this is a two-part problem:

- Disk format, and;
- Interpreting the different physical electrical interface (ID20 rather than the usual ID34).

Is there anything in the works, or a proto-type firmware available?

---

FWIW, I have raised a question on the Retrocomputing site of Stack Exchange regarding the glue logic required for the physical interface and cabling changes, see [Which wires need to be changed to use a Gotek floppy emulator on an Apple Mac?](https://retrocomputing.stackexchange.com/q/25946/202). 

I was hoping that there might be a discrete logic solution to translate the Apple's electrical signals to those of a standard floppy drive, which the Gotek could then be conected to. That way the firmware would not need to be changed when moving the Gotek from use on a BBC to a Macintosh or Apple II, and vice versa. 

However, after some thought, it seems that it would be a bit of a complex job to implement with discrete logic on an external PCB, and that it would be easier (relatively speaking) to just do the whole thing in firmware. This would mean that the Gotek would require reflashing depending upon which platform it was connected to.

There is a video of a HxC prototype which *seemingly* accomplishes this (on an Apple IIc), [Slim HxC Floppy Emulator on Apple 2c](https://www.youtube.com/watch?v=MaR9noqb0JE), so I was wondering if you might be able to as well.

keirf, 2022-12-31:
Mainly it will be an issue of connecting signals correctly to the ID34. However, the head stepper is directly controlled by 4 phase lines PH0-3. So the main work is in connecting those to appropriate input lines on the Gotek, and using them to step between tracks (and half tracks? and quarter tracks?).

Beyond that I reckon you could use HFE image files as a raw container for Apple GCR bitcells.

keirf, 2022-12-31:
Further, I think /WREQ and WR would connect to ID34 Write Gate and Write Data.

RD would connect to ID34 Read Data.

/DRVEN would connect to ID34 Select (pin 10, say)

WRPROT would connect to ID34 Write Protect.

PH0-3 would connect to remaining input pins: MOTOR, DIR, STEP, SIDE. And firmware would need to interpret phase signals on these pins.

I'm not sure on "sense" of some Apple pins (eg is RD active high or active low) and on details like whether phase signals interact with drive select. Nor how/when Apple drive spins up since there's no MOTOR signal.

Probably some work with a scope or logic analyser on a real Drive II will be needed at some points.

drdpj, 2022-12-31:
This might be helpful:
https://embeddedmicro.weebly.com/apple-2iie.html

keirf, 2022-12-31:
Thanks. Ultimately what is needed is someone electronics savvy to actually hook up a Gotek to an Apple II and prepared to run test firmwares and help diagnose problems. Without that, it won't happen.

drdpj, 2022-12-31:
I am willing to risk putting 12v through my gotek and to make up a cable! 😂

keirf, 2022-12-31:
If you hook one up and let me know what's connected where, I can look into doing firmware. It looks very much like signals are mostly active-high. That's easy. And polling the phase signals should be easy too, since they appear to be asserted for many milliseconds each.

Do you happen to know if quarter- and half-tracks are important? I think a "half track" (that is two phases of the four) is a track step and really what is interesting, except for copy protections?

drdpj, 2022-12-31:
I think the half/quarter tracks were used as protection on some disks. The
definitive format to support is "woz" which is used by applesauce.

d.

danieltwagner, 2024-10-23:
I'd also be interested in this and have a classic Mac with working SuperDrive and a Gotek running FlashFloppy. @keirf given that this issue is still open, does that mean you'd still be up for creating the firmware for it? I suppose the first thing would be to wiring up a 20-pin to 34-pin cable, making sure not to fry anything in the process, and then flashing a firmware that'll produce some logs for you?