keirf/flashfloppy #744: FlashFloppy OLED screen resetting when FF.CFG present on USB

craziazkowboi, 2023-01-12:
Hi,

Hoping to get some leads or help - I had 'upgraded' to HxC and have returned to FlashFloppy by re-programming my AT4.35 GOTEK Drive with v3.38 firmware (which is what it was originally).  I used the following instructions to re-program my GOTEK from a Raspberry Pi:
https://github.com/keirf/flashfloppy/discussions/698#discussioncomment-3901428

The update completed successfully and I'm back on FlashFloppy, but not quite 

a) With FF.CFG present on the root of the USB, I see the initial "FlashFloppy+ 3.38 384Kb" screen, but it appears to be resetting endlessly - shows the test, goes blank, shows the test again - flashing on and off every 1 second.

b) If I do not put FF.CFG on the USB drive, the GOTEK starts and does show the first folder, but as soon as I rotate off that folder, the GOTEK spins through the files endlessly, even when spinning the Rotary dial the opposite way - it won't stay still.

I have tried doing a USB Flash update, but it freezes on "FF Update Flash (USB)" either with or without FF.CFG present on the USB drive.   have also tried flashing to 3.37, same result - freezes on the same screen.

I have tried three USB Keys, all formatted to FAT32, and all were working with eh GOTEK Drive prior to trying HxC.

I thought perhaps HxC has written some config which is upsetting FlashFloppy starting so I factory reset my config many times now - no difference.

I have checked that I do not have the programming jumper left on and re-programmed the Drive again from a Raspberry Pi.  When I plug the USB cable in, I get the initial splash screen, "FlashFloppy+ 3.38 384Kb"  and it does not upgrade - either 3.38 or 3.37

Is my GOTEK dead?

Thanks,
Craziazkowboi

keirf, 2023-01-12:
I doubt it's dead. It probably didn't flash properly. Did the programming step include verification? Is there an option to mass erase the whole chip before programming?

The "best" tool is the Artery ISP util on Windows. Of course it needs Windows, isn't signed, and likes to crash on opening. But it does program properly and via a convenient usb-usb cable.

craziazkowboi, 2023-01-12:
Thanks for the reply.  I don’t have a Windows machine unfortunately.  I’m in a Mac and Linux only world, and using dfu-util, I don't know if any of this would help provide a clue or confirm if the whole chip has been erased - or if there is another setting I can use to erase the whole chip?  The help for du-util does not mention an erase option.

$ sudo dfu-util -l 

dfu-util 0.9 

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc. 

Copyright 2010-2016 Tormod Volden and Stefan Schmidt 

This program is Free Software and has ABSOLUTELY NO WARRANTY 

Please report bugs to http://sourceforge.net/p/dfu-util/tickets/ 

Found DFU: [2e3c:df11] ver=0200, devnum=7, cfg=1, intf=0, path="1-1.2", alt=1, name="@Option byte /0x1FFFC000/01*512 g", serial="AT32" 

Found DFU: [2e3c:df11] ver=0200, devnum=7, cfg=1, intf=0, path="1-1.2", alt=0, name="@Internal Flash /0x08000000/512*002Kg", serial="AT32"

Now run dfu-util -D /path/to/flashfloppy-at415-st105-3.36.dfu -a 0 -s 0x08000000:unprotect:force # this will fail with an error, but will unprotect the internal flash in the process.

Here is my result from running dfu-util -D etc as root (using sudo, and noting the '-a 0' is a zero, not an "oh"):

dfu-util -D /path/to/flashfloppy-at415-st105-3.36.dfu -a 0 -s 0x08000000:unprotect:force

$ sudo dfu-util -D flashfloppy-at435-3.38.dfu -a 0 -s 0x08000000:unprotect:force 

dfu-util 0.9 

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc. 

Copyright 2010-2016 Tormod Volden and Stefan Schmidt 

This program is Free Software and has ABSOLUTELY NO WARRANTY 

Please report bugs to http://sourceforge.net/p/dfu-util/tickets/ 

Match vendor ID from file: 0483 

Match product ID from file: df11 

Opening DFU capable USB device... 

ID 2e3c:df11 

Run-time device DFU version 011a 

Claiming USB DFU Interface... 

Setting Alternate Setting #0 ... 

Determining device status: state = dfuIDLE, status = 0 

dfuIDLE, continuing 

DFU mode device DFU version 011a 

Device returned transfer size 2048 

DfuSe interface name: "Internal Flash " 

state(5) = dfuDNLOAD-IDLE, status(0) = No error condition is present 

dfu-util: Wrong state after command "READ_UNPROTECT" downloaddfu-util -D /path/to/flashfloppy-at415-st105-3.36.dfu -a 0

Nearly there, now run dfu-util -D /path/to/flashfloppy-at415-st105-3.36.dfu -a 0 # Now that the flash is unlocked, the device will be programmed.  Again, my results:

$ sudo dfu-util -D flashfloppy-at435-3.38.dfu -a 0 

dfu-util 0.9 

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc. 

Copyright 2010-2016 Tormod Volden and Stefan Schmidt 

This program is Free Software and has ABSOLUTELY NO WARRANTY 

Please report bugs to http://sourceforge.net/p/dfu-util/tickets/ 

Match vendor ID from file: 0483 

Match product ID from file: df11 

Opening DFU capable USB device... 

ID 2e3c:df11 

Run-time device DFU version 011a 

Claiming USB DFU Interface... 

Setting Alternate Setting #0 ... 

Determining device status: state = dfuIDLE, status = 0 

dfuIDLE, continuing 

DFU mode device DFU version 011a 

Device returned transfer size 2048 

DfuSe interface name: "Internal Flash " 

file contains 1 DFU images 

parsing DFU image 1 

image for alternate setting 0, (2 elements, total size = 111964) 

parsing element 1, address = 0x08000000, size = 28960 

Download [=========================] 100% 28960 bytes 

Download done. 

parsing element 2, address = 0x0800c000, size = 82988 

Download [=========================] 100% 82988 bytes 

Download done. 

done parsing DfuSe file

craziazkowboi, 2023-01-12:
I have also tried the drive on the external port, and it behaves the same as internally, checking voltage is tricky as on the Amiga 1000 un less there is a current draw, it doesn't 'activate'.  If I remove the internal floppy power connector (when trying to use internally), the GOTEk OLED stays on though - so Im guessing it is getting some power from the internal floppy connector?

keirf, 2023-01-12:
Yes it will draw power through the 5v pull-up resistors via the gpio pin clamping diodes.

If you have a USB ttl serial adapter you could try programming via stm32flash instead. I have a patch for that somewhere to recognize the mcu id code.

Anyway this definitely all smells like not programming correctly. The problems are q bit subtle to be chip failure. Power supply problems are a possibility I suppose but you can discount this by attaching a stable 5v supply for testing of the menus.

keirf, 2023-01-12:
By the way you don't need to unprotect the mcu. Only factory programming applies flash protection. However in its place you can specify mass-erase and I strongly suggest trying that..

keirf, 2023-01-12:
Another possibility is that the dfu file is bad. The Artery tool uses only the hex file so the dfu file is basically untested. I could try it myself if you continue to have problems.

craziazkowboi, 2023-01-12:
If the DFU were bad (unlikely), I could flash to an older version and that would test that theory right?

Would you know how I do a mass-erase with dfu-util?  I'll try both

Thanks!

keirf, 2023-01-12:
I think others have used the dfu image okay actually. It's probably fine.

I believe you can specify mass-erase instead of unprotect, in the same place on the command line.

craziazkowboi, 2023-01-12:
THANK YOU!!!!   The mass-erase option did it!
I am working again!  HxC has been a bit of nightmare experience!

closed 2023-01-12 by keirf

craziazkowboi, 2023-01-12:
I spoke a bit soon - it works a lot better, but as soon as I put FF.CFG back, it started the ‘winking’ again - from the first splash screen, turning off and back on about every second?

keirf, 2023-01-12:
Upload the FF.CFG, let's have a look.

keirf, 2023-01-12:
This could be perhaps failing to update Flash for some reason.... This could cause endless reset:
1. Read cached config from flash
2. Read FF.CFG from USB
3. Update cached config in Flash (<-- FAILS)
4. Is old cached config different than new cached config in a significant way (eg. display-type has changed)? Reset the Gotek if so.
5. RESET --> Back to step 1.

keirf, 2023-01-12:
Reasons for above could be bad Flash or perhaps some Flash protection (applied by HxC?) which dfu-util has failed to remove. Unfortunately the approach then would be to run the Artery Windows tool. There's some possibility this could work under Wine?

craziazkowboi, 2023-01-12:
> On 12 Jan 2023, at 8:08 pm, Keir Fraser ***@***.***> wrote:
> 
> 
￼￼

Config files attached, thanks!  I have tried do a config reset, but no improvement there yet.

keirf, 2023-01-12:
GitHub strips email attachments. You have to upload via the web interface.

craziazkowboi, 2023-01-12:
[Archive.zip](https://github.com/keirf/flashfloppy/files/10400067/Archive.zip)

FF.CFg and IMG.CFG attached.  Thanks for looking at these.  

I have tried doing a flash update from USB, but it still freezes on that too, dang it!  It sits on "FF Update Flash [USB]"

I used the flash floppy v435 to re-program the GOTEK.

keirf, 2023-01-12:
df-util mass erase and reprogram and see if things generally work before you apply any FF.CFG file. eg does rotary work, does firmware update work, ...

keirf, 2023-01-12:
My best guess is that there's some residual flash sector protection hanging around that neither FlashFloppy nor dfu-util is smart enough to remove. FF in particular assumes that all protections were disabled at programming time. And the only programming tool that knows all about the specifics of 435 option bytes and the protection features they control is... the Artery-supplied tooling.

craziazkowboi, 2023-01-12:
Post the mass-erase and with no FF>CFG file, the rotary does work again, but my Amiga won’t read the disk, it just ignores the drive.  I have a jumper on MO and S0, but removing both (one at a time), still sees the OLED winking.

Removing FF.CFG and replacing with AutoBoot.HFE does see the disk read and I can assign disks to slots, which also work on reboot.  At first I was not getting any of sound from my piezo buzzer, but half-way through playing 9-Fingers it suddenly came to life.  For now I will have to live with this - at least it is working and available again thank you!!

I tried putting FF.CFG back on, and it was still winking, remove it and AutoBoot.HFE works just fine.

I have asked Jeff if he has some code that will cleanup the HxC install.

Thanks again for all the help - rescued thanks to yoU!

Regards,
Craziazkowboi!

craziazkowboi, 2023-01-12:
Once again, you were right.  I erased the GOTEk three times in a row, then re-programmed 3 times in a row - now it working <sigh>  relief!!  Thanks for all the help, could not have done it without you!

craziazkowboi, 2023-01-12:
Anyone interested in my journey to HxC and back, and MacOSX quirks, check out my blog: 
https://amiga1000parceiro.blogspot.com/2022/12/amiga-1000-gotek-drives-mods.html

keirf, 2023-01-12:
Does FlashFloppy OSD on Bluepill really not work with the Artery-based Goteks (or, at least, AT32F435-based Goteks)? It is expected that it should work.

craziazkowboi, 2023-01-12:
Happy to change that comment, I was taking the advice from the GitHub page which referred to older GOTEK models, and my experience with HxC was that if it wasn’t written for a specific model , it would not work - I nearly destroyed my GOTEK  performing the HxC mod only to learn later it wasn’t ready for the 435.  It is now I believe, but that part of my GOTEK is destroyed so I can no longer test.

Regards,
Dwight.

craziazkowboi, 2023-01-12:
<sigh>.  Jeff is upset I called HxC a bit of a nightmare.  Unfortunately it was.  He’ll be really upset if he reads my blog ;)

keirf, 2023-01-12:
Ah I see. All the FF OSD hard stuff is implemented in the Blue Pill, so it is straightforwardly interfaced with any recent version of FlashFloppy on any Gotek.

If there is any confusing text on the FlashFloppy-related wikis do point it out and I'll be happy to reword it.

craziazkowboi, 2023-01-12:
blog updated, thanks.

keirf, 2023-01-12:
Regarding Jeff, he is no lover of me or FlashFloppy, shall we say. I must acknowledge his excellent free and open source software though. The file selector which FlashFloppy also uses (AUTOBOOT.HFE) and especially the HxC software with its unmatched track analyser (visualiser). Many users of Greaseweazle find the HxC software an invaluable companion piece of software.

keirf, 2023-01-12:
> blog updated, thanks.

Thanks, I agree with everything now. OSD built in is a really nice feature I agree. FlashFloppy could support it on the 435 Goteks in future, it is just finding the time to implement it! Stringing a Blue Pill off the Gotek in a custom build is something that DIYers dislike much more than I expected. :)

craziazkowboi, 2023-01-12:
OSD is nice, but emulating 2 x drives from the one GOTEK device would be the the prize-winning feature in my opinion. It’s a feature I would pay for as an add-on if it were available.

keirf, 2023-01-12:
Interesting. I feel it's for sure doable with 435 chip and with OLED and rotary+buttons for UI. You would lose use of MOTOR input signal I think, and would use that for SEL_B instead.

craziazkowboi, 2023-01-12:
Can only speak for myself, but I am not using a jumper on the MOR pins anymore, just S0.  Keeping simple to not confuse my old A1000 :)

craziazkowboi, 2023-01-17:
Hi All,

The mass-erase did allow me to upload a new bootloader and get FlashFloppy working again, but now the only way I can upload is using dfu-util.  If I use the regular method to upgrade, I only get ‘E05' (Flash programming error).  I am unable to upgrade using HxC or FlashFloppy firmware via USB Stick with this current state, but I can upload a .dfu and the Gotek works with everything, except updates.

I’m still learning how these work, and was hoping someone here might know what causes an E05?

Can I mass-erase the alt1 ***@***.*** byte” ROM perhaps?  What do I re-flash/program it with?

Found DFU: [2e3c:df11] ver=0200, devnum=47, cfg=1, intf=0, path="1-1.3", alt=1, ***@***.*** byte   /0x1FFFC000/01*512 g", serial="AT32"
Found DFU: [2e3c:df11] ver=0200, devnum=47, cfg=1, intf=0, path="1-1.3", alt=0, ***@***.*** Flash   /0x08000000/512*002Kg", serial=“AT32"

So far I have been erasing the GOTEK Drive using:

sudo dfu-util -D flashfloppy-at435-3.37.dfu -a 0 -s 0x080000:mass-erase:force

I have also erased using the HxC dfu - just in case the dfu is being used to determine how many bits/bytes to erase?

sudo dfu-util -D gotek_hxc_fw_installer.dfu -a 0 -s 0x08000000:mass-erase:force

This does allow me to upload a new bootstrap firmware, but when I try to update FlashFloppy 3.37 to 3.38, or if I try to update from FlashFloppy 3.37 to a HxC bootloader, I get E05.

If I flash the HxC Boot loader, I get the normal 001 where I need to update the EXCHANGE.BIN file, but when trying to update from here it jumps to 002 and then 880.  I’ve asked Jeff what 880 means as I can’t find a reference.

Any leads/help welcome!

Thanks,

craziazkowboi, 2023-01-18:
I went out and purchased a Windows 11 PC and installed the Artery ISP software.  The short version is, I still get the 880 with HxC and E05 with FlashFloppy.

The Artery ISP erases the drive without a problem, and shows me that all memory locations are unprotected.  It erases, writes the firmware, and verifies 100%.

Is there anything more that can be done?  I do not understand why it would write when re-programmed, but no longer write from a USB update - any advice is welcome.

Thanks,
Dwight.

keirf, 2023-01-18:
You scrolled through the list of all protections and there are no read or write protections at all? There's an unprotect everything option in the programming menu which you might want to select in any case.

Beyond that goodness knows. I have never heard of this issue ever before. New chip time?

craziazkowboi, 2023-01-18:
Confirming I did scroll through the list and checked that everything was set to unprotect.  I also used the unprotect settings near the bottom.  I think it is new chip time.  While totally bizarre that I can write FlashFloppy or HxC using a dfu or hex file using the re-programming method, but I cannot update from USB stick at all.

For now, I’ve flashed 3.38, and will see if I change the drive out in the future, it works for now and replacing is a pain due to how Ive done my mod and mounted it internally.

Thanks very much for the reply though!

Regards,

keirf, 2023-01-18:
Yeah I can only speculate, really. Perhaps page or sector erase is a problem, for example. Whereas mass erase works...

All the while it remains a one-off problem I can't see it being resolved further.