keirf/flashfloppy #747: Akai MPC-2000 Classic wont boot from Boot-disk

andr229k, 2023-01-19:
E:
Gotek with Flashfloppy 3.37 installed.
Jumper = S0
interface = ibmpc-hdout
host = akai

S:
Emulator shows the Boot disk on OLED screen, but does only count up to Track 00.1 as shown on screenshot.

D:
When the boot-disk .HFE is shown on the screen, this indicates that USB-stick is correctly formatted as FAT32.

A:
Tested different USB-sticks sizes and brand, no effect.

Any ideas or suggestions? :)

[image]
[Boot disk 1.72.hfe.txt](https://github.com/keirf/flashfloppy/files/10457608/Boot.disk.1.72.hfe.txt)

andr229k, 2023-01-19:
A:
Updated firmware to 3.38, no effect, still stuck on Track 00.1

keirf, 2023-01-19:
Is W always shown? Indicates writing to disk. Seems unlikely during boot?

andr229k, 2023-01-30:
Faulty Gotek build, device thrown out and new made, problem fixed.

closed 2023-01-30 by andr229k