keirf/flashfloppy #748: Anybody else seeing random read-corruption on Amiga .ADF files?

rjflory, 2023-01-22:
Gotek HW: very-recent Artery AT32F435RG17 SoC.
Gotek FW: 3.37

Target: Two different Amiga A500 systems w/ WB 1.3 .
 
 I have two different Gotek units.  Since I got these a few weeks ago, I've seen random read-corruption at random times at random places.  This has been the source of many hours of debug 'amusement.'..  ;)
 
 I hope this information will be helpful in the debug process.

 - Sometimes programs or text files will be reported as corrupt or will crash the target A500 system(s).
 - Sometimes the A500 target system will ask me to re-insert the Gotek boot-floppy when it has not been removed, written, etc.
 - In one experiment, using the A500 target, I 'cloned' the Gotek boot disk onto four different 'empty' files of a second gotek unit.  
   - Two of the 'clone' files had just one corrupt file in its .adf filesystem.  
   - The directory entries and file-lengths were correct.
   - The corrupt region was exactly 32-bytes in length in both of the "BAD" cases.  Data following the corruption was OK.
   - The corrupt regions were at different offsets into the file(s).
   - In BOTH cases, the corrupt region was a duplication of the preceding 32-bytes of the file. 
   - I tested this using two different Amiga A500 systems (one a Rev-5, the other a Rev-6A).
  
 - I'm using x86 Fedora Linux to access the AFFS .adf images. 
 
ron
embedded software engineer-

keirf, 2023-01-22:
Thanks for the report! What was the method used for cloning? Was the read and write done on the same Gotek?

rjflory, 2023-01-22:
Amiga gui-based "DiskCopy" (basically drag the "WorkBench" boot disk icon onto the "Empty" disk icon).  The operation if read-after-write verify on a per-track basis.

keirf, 2023-01-22:
Thanks for the report! What was the method used for cloning? Was the read and write done on the same Gotek?

I think perhaps a good test will be to hook up to a Greaseweazle and then loop: create a random adf, write to Gotek, read from Gotek, check the random data. That would be close enough, and should detect badness committed to USB drive as there is not enough sram to cache a whole disk. Whereas track by track verify will hit the cache.

keirf, 2023-01-22:
415 has so much testing in the field with no reports, I feel this is probably a bug specific to 435.

rjflory, 2023-01-22:
I added a side-by-side hexdump to original port.
good on left,  bad on right.

rjflory, 2023-01-22:
??? 435 ???   do you mean 4.35?

 is this an internal number.  FW version 3.37 is mentioned on original post.

rjflory, 2023-01-22:
I guest I should try a beta version--  do you recommend a "trusted" (maybe RC) version?

keirf, 2023-01-22:
Thanks. Looks like 64 bytes bad? Interesting that it doesn't seem to be naturally aligned. 

Duplicate of previous 64 bytes... I wonder if it's a USB issue since packet size is 64 bytes.

rjflory, 2023-01-22:
duh...   4 lines of 16 bytes is 64, not 32...

rjflory, 2023-01-22:
yes, in both cases, the miscompare was aligned to xx80 offset in the file.

rjflory, 2023-01-22:
Yes, the possibility of flakey USB drive has been a suspect.  Will have to try some alternate flash drives as well as alternate Gotek FW to narrow down further.

keirf, 2023-01-22:
I'd have to go back over how USB MSC ops get stuffed into packets. There could be an eight-byte header leading straight into data for example. 

Regarding "435", I'm talking about the MCU model AT32F435. The more common model currently is AT32F415 which runs a different firmware build. Did you unexpectedly receive AT32F435? The factory did say that at some point they might switch over to this new more capable chip for their mass production. https://github.com/keirf/flashfloppy/wiki/Gotek-Models#microcontroller-options

keirf, 2023-01-22:
Trying the latest v4.x beta firmware could be interesting as the I/O path is quite different. Although the USB stack is the same.

I wouldn't worry about trying this if it is much bother.

rjflory, 2023-01-22:
In regard to trying 4.x vs 3.x code, are there any known issues regarding upgrade vs downgrade compatability?  I'd hate to brick one...

The person selling these Goteks on ebay touted "uses latest version cpu", so yes, I was aware of some aspect of this and the recent switch-over to new Soc.    ;)

rjflory, 2023-01-22:
BTW, I have noticed that the verify-portion of the Amiga DiskCopy operation will fail at the same place if some slower USB 2.0 flash drives are used.  This would suggest for **this** device, the cache was not hit for the verify.

keirf, 2023-01-22:
You can upgrade and downgrade firmware easily. It's also actually easy to entirely reflash with a usb-a to a or c cable and a Windows PC.

keirf, 2023-01-22:
The problem of failed verify is probably timeout. Currently FlashFloppy will write back to USB before starting to serve read data again. If the write back is slow this could be a problem.

In this regard the v4.x beta firmware is interesting as it writes back only when cylinder changes (when drive heads are stepped). Hence serves reads after writes with much lower and predictable latency.

rjflory, 2023-01-22:
I have upgraded the firmware to 3.37 already, was just asking if there were known incompatibility issues when downgrading from 4.x to earlier 3.x .

keirf, 2023-01-22:
Downgrade will work fine. However you can't go earlier than about 3.34 as that's when support for 435 was added

rjflory, 2023-01-22:
OK, thats exactly the sort of compatibility info I was looking for.

btw, is 4.4a the latest,  links for 4.6a do not appear to be 'live'? (or ready yet)

keirf, 2023-01-23:
4.7a is latest. Link is at top of downloads page on the wiki.

4.6a link does appear to be broken. I will fix it but it's not the latest anyway.

keirf, 2023-01-23:
Just an update: I have been writing ADF images as fast as possible with no verify and no sync to index using Greaseweazle:
```
dd if=/dev/urandom of=a.adf bs=1024 count=880
gw write a.adf --fake-index=300 --no-verify
gw read b.adf
diff a.adf b.adf
```
I haven't found any corruptions so far. This is FF `stable-v3` tip (v3.38 plus a few changes on top) running on a AT32F435 Gotek.

I will say that v3.38 has a 435 specific fix since v3.37. I'm not sure it would make a difference in your setup but it's a possibility.

rjflory, 2023-01-23:
Some thoughts.
- It would be nice to run that verification test over multiple iterations.
- That particular test writes/reads the data as a sequential operation (the least stressful of all scenarios), I wonder if interleaved random seeks to non-sequential tracks/sectors (as one would experience due to target directory and filesystem access) might aggravate problems with the SoC USB-driver, or within the flash drive itself. 
 - Can Greaseweasle do random seeks and read/write 'n' sector-length blocks?  This would allow simple, yet very thorough tests to be scripted and run over extended periods of time.  A nice regression/verification suite...
 
 btw, I just ordered two Greaseweazle units from a Canadian source.  They seem like very handy devices for those of us who occasionally work with, and develop for vintage equipment.  

 It will be nice to have an alternate means of testing/verification the Gotek & HXC drive emulators.

keirf, 2023-01-23:
Random seek would be a good stress test, however I focused on errors introduced by diskcopy, which is a sequential operation (except for disk swaps). Greaseweazle does whole track reads and writes. But then so does Amiga, even when accessing single sectors.

I do have another project called flashfloppy-testbed which allows a modified Gotek to test another, and this can do random (IBM format) sector accesses and the like. It gets run fairly routinely. https://github.com/keirf/flashfloppy-testbed

Other debug aids include a debug build of FlashFloppy which emits logs at 3Mbaud on the serial pins at the programming header at the rear of the Gotek. I can use this to observe write latencies to USB, for example.

rjflory, 2023-01-23:
OK, good information.

keirf, 2023-01-23:
Perhaps a good test would be random track reads from a pseudorandom ADF in the testbed. Then independently verify the sector contents based on knowledge of the PRNG seed.

rjflory, 2023-02-20:
I got distracted by other things, but should have left a note.

 Reflashing my Goteks to 4.7a firmware seems to have resolved the write-corruption issues I was seeing.  I've done many diskcopies and other write ops since, and have not seen this problem since.

closed 2023-02-20 by rjflory

keirf, 2023-02-20:
Thanks for the follow up!