keirf/flashfloppy #752: Gotek along with a physical Disk drive...

mattg1974, 2023-01-30:
I have a amstrad ppc an xt with a shugart floppy controller. I have the gotek set to drive A and left a original floppy as Drive B. The gotek works just fine but if I try to Access the B floppy drive the light turns on and there is access but I always get a access error in dos. 
If I just use both original floppy drives they both work just fine and read disks like they did for the past 35 years. 
Has anyone else had such issues? Did I forget to specify something in the ff.cfg?
Thanks...

keirf, 2023-01-30:
How is the Gotek jumpered?

keirf, 2023-02-07:
Any update before I close thise?

mattg1974, 2023-02-08:
I could resolve the issue with the compaq but still the suggart drive gives me a headache....

Jimbob-B, 2023-03-06:
@mattg1974 I assume you are running a recent version of flashfloppy on the Gotek? This sounds very similar to the issues I was seeing with a physical and Gotek drive on a PlusD interface (that I built) on a ZX Spectrum using the shugart setup. The Gotek would work but trying to access the physical disk drive failed after it read a couple of the tracks. Copying from the Gotek image to a physical disk was impossible. I updated to Flashfloppy 3.38 and this fixed the issues I was having and the setup is working fine now. I don't think I have anything too special in my FF.CFG apart from the shugart settings.

remery66, 2023-03-13:
@keirf I just wanted to report that I was having a similar issue with a new Gotek paired with a 3.5" floppy drive...
My 3.5 would work fine until I accessed the Gotek, and then the floppy would stop working (I/O errors).
I updated my firmware from 3.34 to 3.39 and now it's working! Thank you!
Here is a video I made showing the problem... https://youtu.be/Aa4BXwQKIzE

Cheers!

keirf, 2023-03-13:
Okay yes, this was a bug in the firmware for MCU model AT32F435 (PCB marked SFRKC30.AT4.35).

It affected multi-drive systems only, because the firmware would fail to de-assert the RDATA line when the Gotek is de-selected.

This was fixed in firmware v3.38.