keirf/flashfloppy #753: USB MSC: Detect write cache and issue SYNCHRONIZE_CACHE?

keirf, 2023-02-07:
USB MSC devices may implement write caching and require flushing on sync/close. Should investigate this:
1. Find a USB drive that advertises a write cache
2. Implement a write test that exhibits write loss in the absence of SYNCHRONIZE_CACHE
3. Implement SYNCHRONIZE_CACHE and fix the test

Note this doesn't seem to be a problem for run-of-the-mill USB sticks. Probably no write-back cache.

Leo-PL, 2023-05-28:
Probably any run-off-the-mill SATA SSD in an USB enclosure can help you with that.

Certain (but very rare) USB flash drives with internal USB-SATA bridge and SATA SSD-on-chip can have write cache as well. These are the ones of faster and more expensive variety. Maybe in such cases, the caching may be disabled explicitly?

keirf, 2023-05-28:
Thanks yes disabling the cache might be easiest.