keirf/flashfloppy #754: OAP trying to use his old Amigas again..

ceedy1950, 2023-02-08:
I recently got a gotek from Amazon to play with ..🙄   model is SFRKC30.AT2 for use on one of my Amiga 1200's
Installed artery drivers , used USB A to A lead , windoze device manager only sees the Gotek if the 1&2 pins have a link in them. 
Then seen as a At32 device. 
Artery software loads in OK and then go through  until selecting the flash floppy hex file, then it gives that error in the screen grab. picture 
As I've changed now the write protection it seems to have erased the device ? as the Display is not longer on.
Left it plugged in for a fair while and it gets very hot 😳, tried again this morning and it won't even get seen in Device manager this time ? .
Help anyone ? 

Cheers 

Chris

[image]

keirf, 2023-02-08:
It sounds like you've killed it somehow. It should always be detectable by Windows when jumpered for programming. What gets hot? The tiny MCU on the Gotek PCB?

Your original problem was probably using the wrong HEX file. You need to `flashfloppy-at415-st105-3.38.hex`

ceedy1950, 2023-02-08:
Hi,  thanks for answering so quick .
Seems to be hot around the MCU.   
Only ever used the USB A2A  lead.  and  never  got as far as writing any  Hex file.  as it always gave that error.   so by using the disable write protect, as part of the process has wiped the board ?  and  nothing is recoverable ? 

C

tenuke, 2023-02-08:
Doubt a standard hotel would have something you would want to recover
Nothing is recoverable when the microcontroller is burning.
If you have good USB cable that does not have + and - backwards, the gotek
supposedly had something wrong with it

ceedy1950, 2023-02-08:
found another on Amazon on offer . and its gray this time rather than black..  should arrive Friday.. 
be back  for help then  LOL..

ceedy1950, 2023-02-08:
> 

the A to A USB lead is one I use with my External Hard drive and it still works ok with that,

keirf, 2023-02-08:
Goteks are usually pretty hardy. Unusual for one to burn up. Must be a duff one

keirf, 2023-02-09:
USB USB cable is easiest really. If the cable is known good I'd do same again.

ceedy1950, 2023-02-10:
another gotek arrived,  this is the IC on the board  
[image]

it does appear in device manager  with programming link inserted. 

off for a couple of days  be back soon .. hoping for assistance

keirf, 2023-02-10:
The 48-pin one. Slightly preferable to the 32-pin one, though not much in it. Let us know when you're ready to program it. It should really just be follow the steps in the wiki, and make sure you select the correct HEX file!!

ceedy1950, 2023-02-12:
seems this GOTEK, is a bit different to the last one, says GOTEK on the front  the other had no name form another supplier ?, 
The problem one showed the display all the time,  this new one shows the display when just USB powered , but the display switches off when the programming link is in place .?. but  back home and ready to  try again  LOL.

keirf, 2023-02-12:
This is correct behaviour. You will have no display when jumpered in programming mode .

ceedy1950, 2023-02-12:
so i use the top hex file of the 2 ?  
    AT32F415 MCU: flashfloppy-at415-st105-3.38.hex
    AT32F435 MCU: flashfloppy-at435-3.38.hex....

keirf, 2023-02-12:
Yes that's right

ceedy1950, 2023-02-12:
global erase ?

ceedy1950, 2023-02-12:
[image]

keirf, 2023-02-12:
That's right. Disabling read protection requires mass erase. It's fine.

ceedy1950, 2023-02-12:
looks like  a success ? . display shows F-F  ran 032 uer 3 38 ?

keirf, 2023-02-12:
Yes that is success!

ceedy1950, 2023-02-12:
yippee...  many thanks for the many replies , and the help..

closed 2023-04-21 by keirf