keirf/flashfloppy #755: Sending buttons commands over i2c

sybeck2k, 2023-02-09:
I'd like to be able to send i2c commands to FlashFloppy. I see that right now this is embedded into the OSD code. Did someone implemented already a separated lib for accepting i2c commands? Otherwise, can you give me some pointers on how to include this new functionality?

Silveriomrs, 2023-10-15:
Hi, about that idea... I'm working in a different option... with an Arduino Nano like interface between the Gotek and the controls (rotatory enconder & oLED screen).

philpem, 2024-08-23:
@keirf May I ask what the resolution for this was?

I've been watching it as I've got an Acorn A4000 with an I2C bridge and LCD/touchscreen controller. I was contemplating modifying FF to push display updates to the LCD controller, and read/write to/from from some kind of memory on it to implement a kind of "mailbox" between the A4000 CPU and FF.

End goal was to have a module on the A4000 which switched discs on the FF automatically and ack'ed the "insert disc N" messages.

keirf, 2024-08-23:
No resolution. This was just a bug scrub, and I don't see this one happening any time soon.

Regarding disk autoswap, this can be done via the disk interface if you can step to cylinder 255. That's how autoswap on Amiga is implemented.