keirf/flashfloppy #757: Gotek Artery Controller AT32F415 vs AT32F435

Tallguy297, 2023-02-11:
I have a Commodore Amiga 600, with a Vampire MantiCore V4 accelerator.
There is a problem using the floppy drive with this accelerator using certain Apollo MantiCore firmware.
The firmware that was originally installed on the accelerator card allowed for floppy read & write access.
However this was only made possible because of a jumper setting on the MantiCore board.
This jumper setting allowed you to change from ECS to AGA mode.
When the jumper was inserted the Amiga was in ECS mode and the floppy drive acted as per normal.
Removing the jumper gave the Amiga AGA graphics but disabled the floppy drive.

The Apollo team were made aware of this and they have attempted, with the release of quite a few beta firmwares, to fix this issue.
The latest firmware allows for 6MB AGA Chip memory and floppy drive access without the need to use any jumper settings.
However this does not seem to work very well and the floppy drive read & write access is disabled.

Note: When I mentioned "disabled", the Amiga attempts to read any ADF image but fails with an DF0:Unreadable message on the workbench. If attempting to boot from an ADF or AUTOBOOT.HFE, the drive clicks rapidly for about 5-6 times and stops.

The main reason for writing this to you is that another Apollo team member has a Commodore Amiga 600 with the Gotek floppy drive. He states that with the new MantiCore firmware there is no problem and that the floppy drive reads & writes perfectly.
Other Commodore Amiga 600 Manticore users, like myself, are not having the same success.

So my question is this…
Is there a difference in the way the Gotek Floppy Emulator behaves dependent on what Artery chip is on the main board?
The chip on my Gotek is…<br>
[image]<br>
I have the latest 3.38 firmware flashed on my Gotek.
Thank you...

keirf, 2023-02-11:
Is this an issue with all drives (you refer to "drives" quite generically) or specifically with Gotek drives?

Do you know what Gotek chip the other Apollo team member has in their working Gotek?

In short 415 vs 435 certainly can make a difference as the 435 runs at least 4x faster. I have new builds of FlashFloppy to double the speed of 415 chips however (making them only half the speed of 435, rather than quarter the speed).

keirf, 2023-02-11:
I am also open to working with Apollo on issues with FF or ATK, though I've never been approached.

Tallguy297, 2023-02-11:
> Is this an issue with all drives (you refer to "drives" quite generically) or specifically with Gotek drives?
**I think this is an issue with only Gotek drives, however a few users have genuine floppy drives and encountered the same problems.**
> 
> Do you know what Gotek chip the other Apollo team member has in their working Gotek?
**The member known as Tuko seems to have a working configuration.
I asked the question about the Artery chip in his Gotek flash floppy. I have yet to receive a reply.**
> 
> In short 415 vs 435 certainly can make a difference as the 435 runs at least 4x faster. I have new builds of FlashFloppy to double the speed of 415 chips however (making them only half the speed of 435, rather than quarter the speed).

divined2004, 2023-02-12:
Hello

I just upgraded to latest version of Apollo Icedrake (V4) accelerator and have been experiencing many issues with my Gotek drive. To list some:
- AUTOBOOT is not working any longer. The only way to use my Gotek is via Native mode. However...
- Games booting from the Gotek have audio/video issues out of sync, while loading.
- Graphics glitches which did not exist in previous version.
- 
These issues just appeared when upgrading to latest version of Icedrake.
This is my Gotek chip:

[image]

divined2004, 2023-02-12:
FYI: Reverting to the previous version of the Icedrake core resolved all issues.

Tallguy297, 2023-02-13:
> Is this an issue with all drives (you refer to "drives" quite generically) or specifically with Gotek drives?
>
> Do you know what Gotek chip the other Apollo team member has in their working Gotek? [...]

**Could I test one of these new builds please?**

Tallguy297, 2023-02-13:
> **Could I test one of these new builds please?**

**According to Tuko, he has the STM32F105 chip on his Gotek.
What is the difference between the STM32F105 and the AT32F415?**

keirf, 2023-02-13:
Difference is amount of RAM. STM32F105 has twice as much ram as the 415 chip.

Doubling speed of the 415 may help anyway. I will find you a build to try.

keirf, 2023-02-18:
The following build runs the 415 chip at double speed. Use the non-debug update file. You need to be logged in to GitHub for the download link to work.

https://github.com/keirf/flashfloppy/suites/10889577586/artifacts/549717340

divined2004, 2023-02-18:
Thanks for the link. 
I burned this version and it seems to be working fine with my STM32F105 chip.
However, I am still running the 8898_R9 core of IOcedrake to which I reverted last week. I will test this after updating to Icedrake core 9234 and get back to you.
One more thing, what build is exactly this version you sent? I mean will this functionality be incorporated eventually in the 4.x development version?

keirf, 2023-02-18:
This will be in next 3.x stable release. The 4.x may not stabilise, we will see.

divined2004, 2023-02-18:
I just tested with the Icedrake (Vampire V4) using latest core 9234 and I am unfortunately having the same issues when loading a game from Gotek drive.
- Graphics/audio out of sync.
- Garbled graphics.

keirf, 2023-02-19:
I don't think there's any issue here that FlashFloppy can be directly implicated in. The double speed 415 chip firmware is linked above in case @Tallguy297 wants to try it. Apart from that I don't have anything to do here.

closed 2023-02-19 by keirf

Tallguy297, 2023-02-21:
Thanks for letting my try the new firmware.
I didn't notice any difference from the 3.38 firmware.
I'll wait until the Apollo team figures out this issue.